septentrio-gnss/septentrio_gnss_driver #27: Issue with Serial

Jamalhazem98, 2020-10-26:
Greetings,

I've tried the driver with Raspberry Pi Serial and it's not working well.
I believe it's a problem with receiving/parsing SBF bytes. Sometimes it parses small byte chunks (like 1 byte).
The problem doesn't happen with USB serial. Maybe because it's virtual. 

Thanks a lot

maxpolzin, 2021-11-09:
I am experiencing the same. If I connect via COM1. The logger shows:
"Without a (circular) buffer and a new thread, we would have faced an incomplete message right now. Read unsuccessful: Rx decoder error for message with ID $GPGGA"

Apt package for Ubuntu 20.04, i.e. ROS Noetic, v1.06.

maxpolzin, 2021-11-09:
Compiling v1.0.8 from source on Ubuntu 20.04, ROS Noetic does not print the error anymore, but it shows a similar error message and no messages are being published.

"[DEBUG] [1636460553.012298132]: Calling read_callback_() method, with number of bytes to be parsed being 3"

thomasemter, 2022-03-26:
Hi,

What messages did you activate and what baud rate did you set? I sounds like it could be a problem of a too low interface bandwidth.