septentrio-gnss/septentrio_gnss_driver #51: [BUG] /extsensormeas node is broken

robduer, 2022-04-19:
**Hardware:**
I am using the Septentrio AsteRx-i3 D Pro + on Ubuntu 18.04 and ROS Melodic.

**Bug**
The /extsensormeas node sends the correct angular rate but sends the exact same data for its accelerations. 
Velocity, std_dev and temperature data are way off the real values.
IN RXTools Message Inspector View the correct data on acceleration, angular rate and tempreratur are shown.

**Example**
header: 
  seq: 7638
  stamp: 
    secs: 1650352094
    nsecs:   7669758
  frame_id: "ins"
block_header: 
  sync_1: 36
  sync_2: 64
  crc: 14449
  id: 4050
  length: 100
  tow: 198512000
  wnc: 2206
n: 3
sb_length: 28
source: 32
sensor_model: 10
type: 1
obs_info: 1
**acceleration_X: -0.0325**
acceleration_Y: 0.47875
acceleration_Z: -0.08875
**angular_rate_X: -0.0325**
angular_rate_Y: 0.47875
angular_rate_Z: -0.08875
velocity_X: 9.12120139543e-33
velocity_Y: -1.25499999523
velocity_Z: 9.12119919138e-33
std_dev_X: 1.73937499523
std_dev_Y: -3.21864213941e+26
std_dev_Z: -1.42749989033
sensor_temperature: 28832
zero_velocity_flag: -0.0325

header: 
  seq: 7639
  stamp: 
    secs: 1650352094
    nsecs: 507891427
  frame_id: "ins"
block_header: 
  sync_1: 36
  sync_2: 64
  crc: 63172
  id: 4050
  length: 72
  tow: 198512500
  wnc: 2206
n: 2
sb_length: 28
source: 32
sensor_model: 10
type: 1
obs_info: 1
acceleration_X: 0.01125
acceleration_Y: 0.47125
acceleration_Z: -0.15625
angular_rate_X: 0.01125
angular_rate_Y: 0.47125
angular_rate_Z: -0.15625
velocity_X: 4.05647994111e+29
velocity_Y: 1.05499994755
velocity_Z: -71.6799468994
std_dev_X: 1.7356249094
std_dev_Y: nan
std_dev_Z: -1.53124988079
sensor_temperature: -10491
zero_velocity_flag: 0.01125

thomasemter, 2022-04-22:
Hi there , parsing of /extsensormeas and some other messages is indeed broken. This is fixed in the upcoming release which should be available quite soon. If you need /extsensormeas urgently you might want to try the /dev branch. For the latest fixes also include pull request #49.

robduer, 2022-05-03:
Since updating to 1.1 I cant start the rosnode anymore. 
trying to connect through tcp gives me:

/opt/ros/melodic/share/septentrio_gnss_driver$ roslaunch septentrio_gnss_driver rover.launch param_file_name:=rover
... logging to /home/anon/.ros/log/4ace220c-caaa-11ec-9c2c-902e16fedec2/roslaunch-Legion-15546.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Legion:43073/

SUMMARY
========

CLEAR PARAMETERS
 * /septentrio_gnss/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /septentrio_gnss/activate_debug_log: False
 * /septentrio_gnss/ant_aux1_serial_nr: Unknown
 * /septentrio_gnss/ant_aux1_type: Unknown
 * /septentrio_gnss/ant_serial_nr: Unknown
 * /septentrio_gnss/ant_type: Unknown
 * /septentrio_gnss/att_offset/heading: 0.0
 * /septentrio_gnss/att_offset/pitch: 0.0
 * /septentrio_gnss/aux1_frame_id: aux1
 * /septentrio_gnss/datum: ETRS89
 * /septentrio_gnss/device: tcp://192.168.3.1...
 * /septentrio_gnss/frame_id: gnss
 * /septentrio_gnss/get_spatial_config_from_tf: False
 * /septentrio_gnss/imu_frame_id: imu
 * /septentrio_gnss/ins_initial_heading: auto
 * /septentrio_gnss/ins_spatial_config/ant_lever_arm/x: 0.0
 * /septentrio_gnss/ins_spatial_config/ant_lever_arm/y: 0.0
 * /septentrio_gnss/ins_spatial_config/ant_lever_arm/z: 0.0
 * /septentrio_gnss/ins_spatial_config/imu_orientation/theta_x: 0.0
 * /septentrio_gnss/ins_spatial_config/imu_orientation/theta_y: 0.0
 * /septentrio_gnss/ins_spatial_config/imu_orientation/theta_z: 0.0
 * /septentrio_gnss/ins_spatial_config/poi_lever_arm/delta_x: 0.0
 * /septentrio_gnss/ins_spatial_config/poi_lever_arm/delta_y: 0.0
 * /septentrio_gnss/ins_spatial_config/poi_lever_arm/delta_z: 0.0
 * /septentrio_gnss/ins_spatial_config/vel_sensor_lever_arm/vsm_x: 0.0
 * /septentrio_gnss/ins_spatial_config/vel_sensor_lever_arm/vsm_y: 0.0
 * /septentrio_gnss/ins_spatial_config/vel_sensor_lever_arm/vsm_z: 0.0
 * /septentrio_gnss/ins_std_dev_mask/att_std_dev: 5.0
 * /septentrio_gnss/ins_std_dev_mask/pos_std_dev: 10.0
 * /septentrio_gnss/ins_use_poi: False
 * /septentrio_gnss/leap_seconds: 18
 * /septentrio_gnss/lock_utm_zone: True
 * /septentrio_gnss/multi_antenna: False
 * /septentrio_gnss/ntrip_settings/caster: 0
 * /septentrio_gnss/ntrip_settings/caster_port: 0
 * /septentrio_gnss/ntrip_settings/mode: False
 * /septentrio_gnss/ntrip_settings/mountpoint: 0
 * /septentrio_gnss/ntrip_settings/ntrip_version: v2
 * /septentrio_gnss/ntrip_settings/password: 0
 * /septentrio_gnss/ntrip_settings/rtcm_version: RTCMv2
 * /septentrio_gnss/ntrip_settings/rx_has_internet: False
 * /septentrio_gnss/ntrip_settings/rx_input_corrections_serial: USB2
 * /septentrio_gnss/ntrip_settings/rx_input_corrections_tcp: 6666
 * /septentrio_gnss/ntrip_settings/send_gga: auto
 * /septentrio_gnss/ntrip_settings/username: 0
 * /septentrio_gnss/poi_frame_id: base_link
 * /septentrio_gnss/poi_to_arp/delta_e: 0.0
 * /septentrio_gnss/poi_to_arp/delta_n: 0.0
 * /septentrio_gnss/poi_to_arp/delta_u: 0.0
 * /septentrio_gnss/polling_period/pvt: 500
 * /septentrio_gnss/polling_period/rest: 500
[... 169 lines cut ...]
NODES
  /
    septentrio_gnss (septentrio_gnss_driver/septentrio_gnss_driver_node)
    tf_aux1 (tf2_ros/static_transform_publisher)
    tf_gnss (tf2_ros/static_transform_publisher)
    tf_imu (tf2_ros/static_transform_publisher)
    tf_vsm (tf2_ros/static_transform_publisher)

auto-starting new master
process[master]: started with pid [15281]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a5bd08aa-caa9-11ec-9c2c-902e16fedec2
process[rosout-1]: started with pid [15292]
started core service [/rosout]
process[tf_imu-2]: started with pid [15300]
process[tf_gnss-3]: started with pid [15301]
process[tf_vsm-4]: started with pid [15306]
process[tf_aux1-5]: started with pid [15312]
process[septentrio_gnss-6]: started with pid [15318]
[ WARN] [1651559052.836614601]: /septentrio_gnss: AttEuler needs multi-antenna receiver. Multi-antenna setting automatically activated. Deactivate publishing of AttEuler if multi-antenna operation is not available.
[ INFO] [1651559052.837826470]: /septentrio_gnss: Connecting serially to device/dev/serial/by-id/usb-Septentrio_Septentrio_USB_Device_3239030-if02, targeted baudrate: 921600
[ INFO] [1651559052.838111279]: /septentrio_gnss: Opened serial port /dev/serial/by-id/usb-Septentrio_Septentrio_USB_Device_3239030-if02
[septentrio_gnss-6] process has died [pid 15318, exit code -11, cmd /opt/ros/melodic/lib/septentrio_gnss_driver/septentrio_gnss_driver_node __name:=septentrio_gnss __log:=/home/anon/.ros/log/a5bd08aa-caa9-11ec-9c2c-902e16fedec2/septentrio_gnss-6.log].
log file: /home/anon/.ros/log/a5bd08aa-caa9-11ec-9c2c-902e16fedec2/septentrio_gnss-6*.log

thomasemter, 2022-05-03:
Hi robduer,

the  "Invalid command!" just means that you did not specify your antennas, which means the receiver will not be able to perform in the best way possible but should work nonetheless.  Does `rostopic echo /poscovgeodetic` produce any output with TCP? In any case, thank you for the report, a more meaningful message should be output in this case.

Regarding the serial connection it is very hard to give a diagnosis. Can you please start it with `activate_debug_log: true`? Also,  I must admit that I personally did not test the serial connection thoroughly because TCP over USB gives much more bandwidth.

Cheers,
Thomas

robduer, 2022-05-03:
Hi Thomas,

TCP seems to work now and /extsensormeas is giving the correct accelerations but the sensor_temperature is only sending a seemingly correct message once out of 10 at 10Hz.

By the way, is it possible to change the SBF messagerate inside the .yaml or is it only possible with the RXTool?

header: 
  seq: 487
  stamp: 
    secs: 1651565084
    nsecs:   9079003
  frame_id: "imu"
block_header: 
  sync_1: 36
  sync_2: 64
  crc: 38742
  id: 4050
  revision: 0
  length: 100
  tow: 201902000
  wnc: 2208
n: 3
sb_length: 28
source: [32, 32, 32]
sensor_model: [10, 10, 10]
type: [3, 0, 1]
obs_info: [1, 1, 1]
acceleration_x: -1.1594625
acceleration_y: 3.8206525
acceleration_z: -8.99493
angular_rate_x: -0.0325
angular_rate_y: 0.5075
angular_rate_z: -0.0275
velocity_x: nan
velocity_y: nan
velocity_z: nan
std_dev_x: nan
std_dev_y: nan
std_dev_z: nan
sensor_temperature: 42.7900009155
zero_velocity_flag: nan
---
header: 
  seq: 488
  stamp: 
    secs: 1651565084
    nsecs: 109067120
    frame_id: "imu"
block_header: 
  sync_1: 36
  sync_2: 64
  crc: 4761
  id: 4050
  revision: 0
  length: 72
  tow: 201902100
  wnc: 2208
n: 2
sb_length: 28
source: [32, 32]
sensor_model: [10, 10]
type: [0, 1]
obs_info: [1, 1]
acceleration_x: -1.16424
acceleration_y: 3.8084025
acceleration_z: -8.967
angular_rate_x: -0.05875
angular_rate_y: 0.39875
angular_rate_z: -0.0775
velocity_x: nan
velocity_y: nan
velocity_z: nan
std_dev_x: nan
std_dev_y: nan
std_dev_z: nan
sensor_temperature: -32768.0
zero_velocity_flag: nan

Cropped for readability

---
header: 
  seq: 489
  stamp: 
    secs: 1651565084
    nsecs: 209064386
sensor_temperature: -32768.0
---
header: 
  seq: 490
  stamp: 
    secs: 1651565084
    nsecs: 309313010
sensor_temperature: -32768.0
---
header: 
  seq: 491
  stamp: 
    secs: 1651565084
    nsecs: 409179464
sensor_temperature: -32768.0
---
header: 
  seq: 492
  stamp: 
    secs: 1651565084
    nsecs: 509163461
sensor_temperature: -32768.0
---
header: 
  seq: 493
  stamp: 
    secs: 1651565084
    nsecs: 609121337
sensor_temperature: -32768.0
---
header: 
  seq: 494
  stamp: 
    secs: 1651565084
    nsecs: 709011328
sensor_temperature: -32768.0
---
header: 
  seq: 495
  stamp: 
    secs: 1651565084
    nsecs: 809029756
sensor_temperature: -32768.0
---
header: 
  seq: 496
  stamp: 
    secs: 1651565084
    nsecs: 909058730
sensor_temperature: -32768.0
---
header: 
  seq: 497
  stamp: 
    secs: 1651565085
    nsecs:   9455137
sensor_temperature: 42.7900009155
---
header: 
  seq: 498
  stamp: 
    secs: 1651565085
    nsecs: 109052704
sensor_temperature: -32768.0

robduer, 2022-05-03:
Hi Thomas

the tcp connection seems to work now. 
/extsensormeas is sending the correct accelerations but the sensor_temperature is sending a usefull temperature only once every 10 messages at a messagerate of 10Hz.

It is sending :
sensor_temperature: 42.4000015259
and then 9 times 
sensor_temperature: -32768.0

By the way is there a way to change the messagerate other than having to open the RXTool?

Cheers,
robduer

thomasemter, 2022-05-03:
Hi robduer,

thank you for your feedback. I am happy it works now.

The rates have to be set in the .yaml. by `polling_period/pvt` and `polling_period/rest` in ms. Setting it to 0 means "on change", i.e., as fast as possible.

Sensor temperature is not sent in every cycle of the /extsensormsg block and set to its do-not-use-value if it is not present. SBF and ROS messages are somewhat different in this regard. SBF has optional fields while ROS messages are always complete.

Cheers,
Thomas

closed 2022-05-03 by robduer