septentrio-gnss/septentrio_gnss_driver #59: Issues trying to launch rover.launch file

aFriendlyNoob, 2022-05-04:
Hello, 
after following the steps to install the driver and populating the rover.yaml file to the desired values for my ins (AsteRx-i D UAS) i reiceive the following errors when trying to launch using the launch command: roslaunch septentrio_gnss_driver rover.launch param_file_name:=rover.

[ERROR] [1651696982.905807212]: /septentrio_gnss: Invalid command just sent to the Rx! The Rx's response contains 40 bytes and reads:
 $R? sga, MultiAntenna : Invalid command!
[ERROR] [1651696983.164258781]: /septentrio_gnss: Invalid command just sent to the Rx! The Rx's response contains 55 bytes and reads:
 $R? sivl, VSM1, -0.348, 0.546, 0.711 : Invalid command!
[ERROR] [1651696983.288957650]: /septentrio_gnss: Invalid command just sent to the Rx! The Rx's response contains 33 bytes and reads:
 $R? siih, auto : Invalid command!

Tried troubleshooting myself but am not sure what I am missing. Any ideas?

`# Configuration Settings for the Rover Rx

# GNSS/INS Parameters

device: tcp://192.168.3.1:28784

serial:
  baudrate: 115200
  rx_serial_port: USB0
  hw_flow_control: off

frame_id: gnss

imu_frame_id: imu

poi_frame_id: base_link

vsm_frame_id: vsm

aux1_frame_id: aux1

vehicle_frame_id: base_link

get_spatial_config_from_tf: false

lock_utm_zone: true

use_ros_axis_orientation: true

receiver_type: ins

datum: ETRS89

poi_to_arp:
  delta_e: 0.0
  delta_n: 0.0
  delta_u: 0.0

att_offset:
  heading: 0.0
  pitch: 0.0
  
ant_type: Unknown
ant_serial_nr: Unknown
ant_aux1_type: Unknown
ant_aux1_serial_nr: Unknown

leap_seconds: 18

polling_period:
  pvt: 500
  rest: 500

use_gnss_time: false

ntrip_settings:
  mode: off
  caster: 0
  caster_port: 0
  username: 0
  password: 0
  mountpoint: 0
  ntrip_version: v2
  send_gga: auto
  rx_has_internet: false
  rtcm_version: RTCMv2
  rx_input_corrections_tcp: 6666
  rx_input_corrections_serial: USB2

publish:
  # For both GNSS and INS Rxs
  navsatfix: false
  gpsfix: true
  gpgga: false
  gprmc: false
  gpst : false
  measepoch: false
  pvtcartesian: false
  pvtgeodetic: true
  poscovcartesian: false
  poscovgeodetic: true
  velcovgeodetic: true
  atteuler: true
  attcoveuler: false
  pose: true
  diagnostics: false
  # For GNSS Rx only
  gpgsa: false
  gpgsv: false
  # For INS Rx only
  insnavcart: false
  insnavgeod: false
  extsensormeas: false
  imusetup: true
  velsensorsetup: false
  exteventinsnavcart: false
  exteventinsnavgeod: false
  imu: false
  localization: false
  tf: false

# INS-Specific Parameters

ins_spatial_config:
  imu_orientation:
    theta_x: 180.0
    theta_y: 0.0
    theta_z: 180.0
  poi_lever_arm:
    delta_x: 0.0
    delta_y: 0.0
    delta_z: 0.0
  ant_lever_arm:
    x: -0.348
    y: -0.546
    z: -0.711
  vel_sensor_lever_arm:
    vsm_x: 0.0
    vsm_y: 0.0
    vsm_z: 0.0

ins_initial_heading: auto

ins_std_dev_mask:
  att_std_dev: 5.0
  pos_std_dev: 10.0

ins_use_poi: false

# Logger

activate_debug_log: false`

thomasemter, 2022-05-05:
Hi @aFriendlyNoob,

the driver was developed with latest generation of Septentrio receivers. The AsteRx-i D UAS has an older firmware and does not support all new features like velocity aiding. Despite the error messages the driver should output valid data. Can you confirm this?

After a quick look into the firmware reference guide, I can confirm that the three errors you get are indeed not supported.

- Newer receivers let you choose to use multi-antenna for heading aiding. As I read the guide, your receiver automatically uses multi-antenna if it has the appropriate permission. Just make sure to set the correct att_offset for the secondary antenna.
-  The AsteRx-i D UAS does not support velocity aiding, so no frame can be set.
- `siih` corresponds to the parameter `ins_initial_heading`, it determines if the heading is initialized from GNSS or from a stored value. This is not supported by the AsteRx-i D UAS.

aFriendlyNoob, 2022-05-09:
> Hi @aFriendlyNoob,
>
> the driver was developed with latest generation of Septentrio receivers. The AsteRx-i D UAS has an older firmware and does not support all new features like velocity aiding. [...]

Thank you for the detailed response, after launching the rover launch file with

aFriendlyNoob, 2022-05-10:
> Hi @aFriendlyNoob,
>
> the driver was developed with latest generation of Septentrio receivers. The AsteRx-i D UAS has an older firmware and does not support all new features like velocity aiding. [...]

Thank you for such a detailed response, the errors seemed to have been cleared after messing around with it a bit and it is outputting valid data but am failing to get it to output the orientation. Any ideas? I will attach my updated rover.yaml `file.`# Configuration Settings for the Rover Rx

# GNSS/INS Parameters

device: tcp://192.168.3.1:28784
#device: serial:/dev/ttyACM1

serial:
  baudrate: 115200
  rx_serial_port: /dev/ttyAMC1
  hw_flow_control: off

frame_id: gnss

imu_frame_id: imu

poi_frame_id: base_link

#vsm_frame_id: vsm

aux1_frame_id: aux1

vehicle_frame_id: base_link

get_spatial_config_from_tf: false

lock_utm_zone: true

use_ros_axis_orientation: true

receiver_type: gnss 

#multi_antenna: true

datum: ETRS89

poi_to_arp:
  delta_e: 0.0
  delta_n: 0.0
  delta_u: 0.0

att_offset:
  heading: 90.0
  pitch: 0.0
  
ant_type: Unknown
ant_serial_nr: Unknown
ant_aux1_type: Unknown
ant_aux1_serial_nr: Unknown

leap_seconds: 18

polling_period:
  pvt: 500
  rest: 500

use_gnss_time: false

ntrip_settings:
  mode: off
  caster: 0
  caster_port: 0
  username: 0
  password: 0
  mountpoint: 0
  ntrip_version: v2
  send_gga: auto
  rx_has_internet: false
  rtcm_version: RTCMv2
  rx_input_corrections_tcp: 6666
  rx_input_corrections_serial: ACM0

publish:
  # For both GNSS and INS Rxs
  navsatfix: true
  gpsfix: true
  gpgga: false
  gprmc: false
  gpst : false
  measepoch: false
  pvtcartesian: false
  pvtgeodetic: true
  poscovcartesian: false
  poscovgeodetic: true
  velcovgeodetic: true
  atteuler: true
  attcoveuler: false
  pose: true
  diagnostics: true
  # For GNSS Rx only
  gpgsa: false
  gpgsv: false
  # For INS Rx only
  insnavcart: false
  insnavgeod: false
  extsensormeas: false
  imusetup: true
  velsensorsetup: false
  exteventinsnavcart: false
  exteventinsnavgeod: false
  imu: true
  localization: true
  tf: false

# INS-Specific Parameters

ins_spatial_config:
  imu_orientation:
    theta_x: 0 
    theta_y: 0.0
    theta_z: 180.0
  poi_lever_arm:
    delta_x: 0 
    delta_y: 0
    delta_z: 0
  ant_lever_arm:
    x: -0.348
    y: -0.546
    z: 0.711

#ins_initial_heading: 

ins_std_dev_mask:
  att_std_dev: 5.000
  pos_std_dev: 100.000

ins_use_poi: false

# Logger

activate_debug_log: false``

thomasemter, 2022-05-11:
You have to set ´receiver_type: ins´ to let the receiver work in INS mode. Then you can get the orientation from e.g. insnavgeod or in the imu topic. You also might want to use the position from insnavgeod instead of pvtgeodetic because the later only incoroporates GNSS.

Mubashir-alam, 2023-04-09:
whenever I run `roslaunch septentrio_gnss_driver rover.launch param_file_name:=rover` I get the following error 

```ERROR: cannot launch node of type [septentrio_gnss_driver/septentrio_gnss_driver_node]: Cannot locate node of type [septentrio_gnss_driver_node] in package [septentrio_gnss_driver]. Make sure file exists in package path and permission is set to executable (chmod +x)```

thomasemter, 2023-04-10:
Did you install it via ROS packages or did you build it from source? In the latter case, have you sourced `setup.bash` of your workspace (cf. [README.md](https://github.com/septentrio-gnss/septentrio_gnss_driver#usage))?

Mubashir-alam, 2023-04-10:
I install it via ROS package as `sudo apt install ros-noetic-septentrio-gnss-driver
`

thomasemter, 2023-04-12:
Hm, that is weird. Are you able to run other ROS nodes? Can you please check if `septentrio_gnss_driver` shows up when you run `rospack list` and if `septentrio_gnss_driver_node` exists under `/opt/ros/noetic/lib/septentrio_gnss_driver`?