septentrio-gnss/septentrio_gnss_driver #61: Best way to integrate global heading (dual antenna) to robot_localization

klausener, 2022-05-12:
Hello together,

I've just installed the mosaic-h receiver to our robot and I wonder what is the best way to integrate the global heading (two antennas) into the robot_localization node. I have found the /pose topic, which would be in general an easy way to integrate (geometry_msgs/PoseWithCovarianceStamped.msg),but it looks like it doesn't include the global but local heading.

Thanks in advance

thomasemter, 2022-05-12:
Hi there,

The heading output by the receiver I indeed a global heading referring to true north. Perhaps the confusion comes from the fact that the attitude is given in a local frame the moves with the vehicle, but this is mostly relevant for roll and pitch. As this local frame is aligned to north and east, the heading is also the global one. I am not familiar with robot_localization but you have to be aware that the global heading is versus true north, which is not in a cartesian frame. For localization in small areas this won’t matter but if you plan to use it for long range or even global level localization this has to be taken into account.

I hope this helps.

klausener, 2022-05-13:
Thank you very much for your answer! I checked the _/pose_ topic and the included orientation. It looks like you've said. The orientation is referring to true north. I was a little bit confused since in the topic description of topic _/pose_ is written:  

> The pose field of this ROS message contains position with respect to the absolute ENU frame (longitude, latitude, height), while the orientation is with respect to a vehicle-fixed (e.g. for mosaic-x5 in moving base mode via the command setAttitudeOffset, ...) !local! NED frame or ENU frame if use_ros_axis_directions is set true. Thus the orientation is !not! given with respect to the same frame as the position is given in. The cross-covariances are hence set to 0.

But I guess this is just the case, if there is no "global" heading available. The global oirentation can thus be very easily integrated into the EKF of the robot_localization via the topic _/pose_!

Thanks!

closed 2022-05-13 by klausener

thomasemter, 2022-05-16:
Sorry to bring this up again. I had a look at robot_localization and as it turns out `/pose` is not well suited after all. Instead a combination of the topics `/imu`  and `/navsatfix` (the latter via [`navsat_transform_node`](https://docs.ros.org/en/api/robot_localization/html/navsat_transform_node.html)) should be considered.

thomasemter, 2022-05-16:
Sorry to bring this up again. I had a look at robot_localization and as it turns out /pose is not well suited after all.

klausener, 2022-05-16:
Yes, you are right. ` /pose` has geodetic coordinates and isn't suitable for robot_localization regarding position. But I was searching for an easy way to include the global orientation of the dual antenna receiver. The ekf of `robot_localization` offers an configuration vector to define which of the information of an incoming message should be used. In my case, I just use the yaw-rotation from this message. The position is fused by the `navsat_transform_node`. Thanks for your additional information.

thomasemter, 2022-05-16:
Perfect, so you already integrated it correctly :-) Thank you for your confirmation.