septentrio-gnss/septentrio_gnss_driver #64: Driver crashing with double free or corruption error

wouter-heerwegh, 2022-06-02:
Hi,

The driver seems to crash if I run it for a longer period of time. It gave a double free or corruption error.

My setup is as follows:
Asterx i3 D Pro+ connected over usb to a pc.
bnc ntrip client connecting on usb2 of the asterx i3.

wouter-heerwegh, 2022-06-02:
Hi @thomasemter,

I'm using the v1.1.1, with ROS noetic. I'm using a usb connection. My configuration is:

```
device: serial:/dev/sep_gps

serial:
  baudrate: 115200
  rx_serial_port: USB1
  hw_flow_control: off

login:
  user: ""
  password: ""

frame_id: gnss

imu_frame_id: imu

poi_frame_id: base_link

vsm_frame_id: vsm

aux1_frame_id: aux1

vehicle_frame_id: base_link

get_spatial_config_from_tf: false

lock_utm_zone: true

use_ros_axis_orientation: true

receiver_type: gnss

multi_antenna: false

datum: ETRS89

poi_to_arp:
  delta_e: 0.0
  delta_n: 0.0
  delta_u: 0.0

att_offset:
  heading: 0.0
  pitch: 0.0
  
ant_type: Unknown
ant_serial_nr: Unknown
ant_aux1_type: Unknown
ant_aux1_serial_nr: Unknown

leap_seconds: 18

polling_period:
  pvt: 500
  rest: 500

use_gnss_time: false

ntrip_settings:
  mode: off
  caster: 0
  caster_port: 0
  username: 0
  password: 0
  mountpoint: 0
  ntrip_version: v2
  send_gga: auto
  rx_has_internet: false
  rtcm_version: RTCMv2
  rx_input_corrections_tcp: 6666
  rx_input_corrections_serial: USB2

publish:
  # For both GNSS and INS Rxs
  navsatfix: true
  gpsfix: true
  gpgga: false
  gprmc: false
  gpst : false
  measepoch: false
  pvtcartesian: false
  pvtgeodetic: true
  poscovcartesian: false
  poscovgeodetic: true
  velcovgeodetic: true
  atteuler: false
  attcoveuler: true
  pose: false
  diagnostics: false
  # For GNSS Rx only
  gpgsa: false
  gpgsv: false
  # For INS Rx only
  insnavcart: false
  insnavgeod: false
  extsensormeas: false
  imusetup: false
  velsensorsetup: false
  exteventinsnavcart: false
  exteventinsnavgeod: false
  imu: false
  localization: false
  tf: false

# INS-Specific Parameters

ins_spatial_config:
  imu_orientation:
    theta_x: 0.0
    theta_y: 0.0
    theta_z: 0.0
  poi_lever_arm:
    delta_x: 0.0
    delta_y: 0.0
    delta_z: 0.0
  ant_lever_arm:
    x: 0.0
    y: 0.0
    z: 0.0
  vel_sensor_lever_arm:
    vsm_x: 0.0
    vsm_y: 0.0
    vsm_z: 0.0

ins_initial_heading: auto

ins_std_dev_mask:
  att_std_dev: 5.0
  pos_std_dev: 10.0

ins_use_poi: false

# Logger

activate_debug_log: false
```

wouter-heerwegh, 2022-06-17:
Changing the baudrate didn't seem to help. I did enable tethering to be able to connect over TCP with the GPS and that seems to work fine.

It would maybe be nice to have a section in the readme dedicated on how to enable RNDIS on linux.

closed 2022-06-17 by wouter-heerwegh

thomasemter, 2022-06-17:
Thank you for your valuable feedback. We have a short explanation on RNDIS in the connection parameters section. Can you please elaborate on what you found missing or did you encounter difficulties? On our systems RNDIS worked right out of the box. 

Also, i did some more digging into the code and I think I have found the bug causing the memory corruption (PR #65 or commit [be50f12](https://github.com/septentrio-gnss/septentrio_gnss_driver/pull/65/commits/be50f12e8a6f9a84e45a58aa9d7889fb3dbfbd7a)). Would it be possible for you to pull this fix and test it again with the serial connection?