septentrio-gnss/septentrio_gnss_driver #69: How not to reflect driver parameter settings in hardware

rsasaki0109, 2022-08-24:
@tibordome @thomasemter 

I use aster rx and use the ros2 branch of this package.

In the process I encountered the following problem.

Even if I set theta_x in imu_orientation to 180, it is 0.0deg (setting values within 90 degrees are reflected correctly).
https://github.com/septentrio-gnss/septentrio_gnss_driver/blob/ros2/config/rover.yaml#L112

aster rx is not able to access the RTK correction service correctly even if I set it according to the following

https://github.com/septentrio-gnss/septentrio_gnss_driver/pull/16/commits/debf0e4b7766fa5ad5554d09ee442f27536aa7f4#diff-b06a267690b8225807484c743113fa3cf06dc111f3a59de9c57f3b9f32c6303d

So can you tell me how to set the settings correctly or how Aster settings are not done from the ROS driver?

thomasemter, 2022-08-26:
Hi @rsasaki0109 ,

if you have set 'use_ros_axis_orientation' to true, the orientation is rotated by 180 ° around the x-axis as Septentrio follows the front-right-down convention, while ROS adopts front-left-up. When using the ROS axis convention, the driver converts between the two conventions and your observation should be correct. In case you want to purely use the convention of the receiver, just set 'use_ros_axis_orientation' to false.

Regarding the access to RTK corrections: If the values are passed correctly to the receiver, the driver seems to work correctly :-). Maybe your receiver does not have access to the internet? Is it also connected via ethernet to a network with a route to the internet?

rsasaki0109, 2022-09-09:
Sorry for the late reply. Thank you for your reply. It's not easy to check due to other tasks, but I will.

thomasemter, 2022-09-09:
You are welcome. Just take your time :-)

rsasaki0109, 2022-09-25:
Thank you very much. I was able to check the operation safely. I will close this issue.

closed 2022-09-25 by rsasaki0109

thomasemter, 2022-09-26:
Great, thank you for your feedback.