septentrio-gnss/septentrio_gnss_driver #71: Wrong heading angle of Mosaic-H module

hangcui1201, 2022-09-29:
Hi, 
We used the septentrio_gnss_driver with mosaic-H module and read the heading angle of ROS topic /atteuler. We found the heading/attitude angle of /atteuler was (1) not consistent with the heading angle from the web gui. (2) wrong at North, the heading angle of main antenna pointing to the aux antenna was 270. Should it be 0 or 360?

For example, taking the vector of the main antenna pointing to the aux antenna,
North: 0 degree
East: 90 degree
South: 180 degree
West: 270 degree

However, the output heading angle of /atteuler was confusing. 

[image]

[image]

Thanks!

thomasemter, 2022-09-30:
Hi,

if you have set `use_ros_axis_orientation` to `true` this is expected behavior.  Septentrio follows the NED (yaw = 0 is north, positive clockwise) convention, whereas ROS is ENU (yaw = 0 is east, positive counterclockwise) and the driver converts between the two. There is no conversion when setting this parameter to `false` and the angles should be consistent.

hangcui1201, 2022-10-01:
Thanks a lot! It worked!

thomasemter, 2022-10-02:
Great, thank you for your feedback.