septentrio-gnss/septentrio_gnss_driver #73: Invalid argument for setting lever arm

Jad-Tawil, 2022-10-06:
The driver is reading frames from tf. The following is the message received from the receiver:

```
[ERROR] [1665063739.518137652]: /septentrio_gnss: Invalid command just sent to the Rx! The Rx's response contains 47 bytes and reads:
 $R? setINSAntLeverArm: Argument 'X' is invalid!
```

I am printing the exact string being formed prior to the sending, which is:

`sial, 0.03639, 0.176597, -0.836319`

Looking at the manual, this looks correct.

Why would the receiver respond this way?

Jad-Tawil, 2022-10-06:
Understood. 
So the receiver does not accept more than 3 decimal places? That means that nothing more precise than millimeters is processed? That seems strange, accuracy is obviously impacted in that situation.

thomasemter, 2022-10-06:
In my opinion, millimeter precision should be adequate for lever arms. We have set up several robots and we were never able to measure them more precisely.

If highest accuracy matters for your application, also make sure to specify your antenna types:
- ant_type: ""
- ant_serial_nr: ""
- ant_aux1_type: ""
- ant_aux1_serial_nr: ""