septentrio-gnss/septentrio_gnss_driver #77: Publish RTCM

arpaterson, 2023-04-04:
Not an issue, but a suggestion.

It would be nice to be able to simply publish and consume [rtcm_msgs](https://github.com/tilk/rtcm_msgs) or similar in some admittedly more obscure set ups.

One such setup is a location with no internet connection, but with 
- one or more rovers (separate ROS instances with receivers), 
- data link (wifi, or other data link, with some method of ROS topic bridging) 
- and a local RTK reference station (another ROS instance, with another receiver and survey antenna).

With the above we can deploy at any site, without the requirement for any internet based NTRIP caster. Instead, using ROS messages to distribute RTCM. This keeps it nice and tidy, with no 'outside ROS' software or communications (eg. u-center, SNIP, which are both non-linux, or even any linux NTRIP caster).

I currently use an F9P, and don't have a Septentrio receiver to trial, unfortunately. 
But this receiver looks promising and I hope to obtain one soon.

thomasemter, 2023-04-12:
Septentrio receivers have built-in NTRIP casters and are able to consume NTRIP streams. This works  over the internet as well as within local networks. So you could have such a setup out of the box without taking a detour via ROS and without the need for any 'outside' software.

arpaterson, 2023-04-16:
> Septentrio receivers have built-in NTRIP casters and are able to consume NTRIP streams. This works over the internet as well as within local networks. So you could have such a setup out of the box without taking a detour via ROS and without the need for any 'outside' software.

That sounds great.
Just to be perfectly clear on terminology: Do they have built in NTRIP *casters* (distribution server that rtcm consuming clients can connect to), or NTRIP *servers* (client that uploads rtcm to an NTRIP caster)?

thomasemter, 2023-04-16:
Septentrio GNSS receivers do have both, a built-in NTRIP caster and NTRIP servers. The caster can be set up to broadcast corrections from the receiver itself (i.e., generated by the receiver’s own NTRIP server) or from any remote NTRIP server.

sukhrajklair, 2024-08-28:
In our scenario, we just have a USB connection to the receiver. It would be nice to pass RTCM corrections to the receiver using the same connection that the driver uses to receive data from the receiver. There are ROS packages that can use NTRIP to receive corrections and publish on a topic.

thomasemter, 2024-08-28:
Hi @sukhrajklair ,

the USB-Connection provides two virtual serial connections, one of which you may configure to receive RTCM corrections.

sukhrajklair, 2024-08-28:
Thanks for your response @thomasemter 
Is there a node in this package that could subscribe to a topic with rtcm data and transmit the data to the receiver?

thomasemter, 2024-08-28:
There is no node in the package for that. Maybe you could use a software like [BKG client](https://igs.bkg.bund.de/ntrip/bnc), which can forward NTRIP to serial.