septentrio-gnss/septentrio_gnss_driver #98: driver does not publish topics

Targacore, 2023-07-26:
When I launch rover.launch with the rover.yaml file the _/septentrio_gnss_ node does only publishes to _/rosout_ and _/tf_ topic. 
I changed the device and adapted the baudrate for my device, the rest is default.
```
device: serial:/dev/ttyACM0

serial:
  baudrate: 115200
  rx_serial_port: ACM0
  hw_flow_control: off
```
The node receives data via Serial port but does not publish any odometry or navsat messages.

```
process[septentrio_gnss-5]: started with pid [2744]
[DEBUG] [1690366183.397064203]: /septentrio_gnss: Called ROSaicNode() constructor..
[DEBUG] [1690366183.645232391]: /septentrio_gnss: IMU roll offset: 180.000000
[DEBUG] [1690366183.645283271]: /septentrio_gnss: IMU pitch offset: -0.000000
[DEBUG] [1690366183.645393205]: /septentrio_gnss: IMU yaw offset: -0.000000
[DEBUG] [1690366183.645429037]: /septentrio_gnss: Ant heading offset: -0.000000
[DEBUG] [1690366183.645457700]: /septentrio_gnss: Ant pitch offset: -0.000000
[ WARN] [1690366183.688190502]: /septentrio_gnss: AttEuler needs multi-antenna receiver. Multi-antenna setting automatically activated. Deactivate publishing of AttEuler if multi-antenna operation is not available.
[DEBUG] [1690366183.688234891]: /septentrio_gnss: Finished getROSParams() method
[DEBUG] [1690366183.688260394]: /septentrio_gnss: Called initializeIO() method
[DEBUG] [1690366183.688375014]: /septentrio_gnss: Searching for serial port/dev/ttyACM0
[DEBUG] [1690366183.688443967]: /septentrio_gnss: Leaving initializeIO() method
[DEBUG] [1690366183.688467093]: /septentrio_gnss: Called defineMessages() method
[DEBUG] [1690366183.688492509]: /septentrio_gnss: Key 4007 successfully inserted into multimap: true
[DEBUG] [1690366183.688517012]: /septentrio_gnss: Key 5906 successfully inserted into multimap: true
[DEBUG] [1690366183.688550000]: /septentrio_gnss: Called connect() method
[DEBUG] [1690366183.688594200]: /septentrio_gnss: Key 5908 successfully inserted into multimap: true
[DEBUG] [1690366183.688636792]: /septentrio_gnss: Key 5938 successfully inserted into multimap: true
[DEBUG] [1690366183.688673906]: /septentrio_gnss: Key 5939 successfully inserted into multimap: true
[DEBUG] [1690366183.688710510]: /septentrio_gnss: Key 4027 successfully inserted into multimap: true
[DEBUG] [1690366183.688751519]: /septentrio_gnss: Key NavSatFix successfully inserted into multimap: true
[DEBUG] [1690366183.688797077]: /septentrio_gnss: Key GPSFix successfully inserted into multimap: true
[DEBUG] [1690366183.688830181]: /septentrio_gnss: Key 4013 successfully inserted into multimap: true
[DEBUG] [1690366183.688869005]: /septentrio_gnss: Key 4001 successfully inserted into multimap: true
[DEBUG] [1690366183.688908102]: /septentrio_gnss: Key DiagnosticArray successfully inserted into multimap: true
[DEBUG] [1690366183.688948936]: /septentrio_gnss: Key 4014 successfully inserted into multimap: true
[DEBUG] [1690366183.689018041]: /septentrio_gnss: Key 4082 successfully inserted into multimap: true
[DEBUG] [1690366183.689061572]: /septentrio_gnss: Key 5902 successfully inserted into multimap: true
[DEBUG] [1690366183.689093404]: /septentrio_gnss: Leaving defineMessages() method
[DEBUG] [1690366183.689121600]: /septentrio_gnss: Called configureRx() method
[DEBUG] [1690366183.689148338]: /septentrio_gnss: Started timer for calling reconnect() method until connection succeeds
[DEBUG] [1690366183.689274291]: /septentrio_gnss: Called reconnect() method
[ INFO] [1690366183.689312854]: /septentrio_gnss: Connecting serially to device/dev/ttyACM0, targeted baudrate: 115200
[DEBUG] [1690366183.689339701]: /septentrio_gnss: Calling initializeSerial() method..
[ INFO] [1690366183.689493281]: /septentrio_gnss: Opened serial port /dev/ttyACM0
[DEBUG] [1690366183.689519236]: /septentrio_gnss: Our boost version is 106501.
[DEBUG] [1690366183.689556710]: /septentrio_gnss: Creating new Async-Manager object..
[DEBUG] [1690366183.689598457]: /septentrio_gnss: Setting the private stream variable of the AsyncManager instance.
[DEBUG] [1690366183.689724192]: /septentrio_gnss: Launching tryParsing() thread..
[DEBUG] [1690366183.689791967]: /septentrio_gnss: Called setManager() method
[DEBUG] [1690366183.689826916]: /septentrio_gnss: Leaving setManager() method
[DEBUG] [1690366183.689849724]: /septentrio_gnss: Gradually increasing the baudrate to the desired value...
[DEBUG] [1690366183.689870545]: /septentrio_gnss: Initiated current_baudrate object...
[DEBUG] [1690366183.689893544]: /septentrio_gnss: Current baudrate is 115200
[ INFO] [1690366183.689914272]: /septentrio_gnss: Set ASIO baudrate to 115200, leaving InitializeSerial() method
[DEBUG] [1690366183.689937896]: /septentrio_gnss: Leaving reconnect() method
[DEBUG] [1690366183.690029536]: /septentrio_gnss: Sent the following 22 bytes to the Rx: 
SSSSSSSSSSSSSSSSSSS
[DEBUG] [1690366183.706319800]: /septentrio_gnss: Calling read_callback_() method, with number of bytes to be parsed being 108
[DEBUG] [1690366183.724224402]: /septentrio_gnss: Calling read_callback_() method, with number of bytes to be parsed being 19
[DEBUG] [1690366183.754251459]: /septentrio_gnss: Calling read_callback_() method, with number of bytes to be parsed being 93
[DEBUG] [1690366183.808427245]: /septentrio_gnss: Calling read_callback_() method, with number of bytes to be parsed being 128
[DEBUG] [1690366183.808860588]: /septentrio_gnss: The NMEA message contains 125 bytes and is ready to be parsed. It reads: $GPGGA,000217.699,,,,,0,0,,,M,,M,,*4A???;
[DEBUG] [1690366183.809268898]: /septentrio_gnss: The NMEA message contains 49 bytes and is ready to be parsed. It reads: $GPGSA,A,1,,,,,,,,,,,,,,,*1E???/
[DEBUG] [1690366183.910394340]: /septentrio_gnss: Calling read_callback_() method, with number of bytes to be parsed being 108
[DEBUG] [1690366183.910556725]: /septentrio_gnss: The NMEA message contains 105 bytes and is ready to be parsed. It reads: $GPGGA,000217.800,,,,,0,0,,,M,,M,,*44I??;
[DEBUG] [1690366183.910746510]: /septentrio_gnss: The NMEA message contains 29 bytes and is ready to be parsed. It reads: $GPGSA,A,1,,,,,,,,,,,,,,,*1E,
[... 62 lines cut ...]
    * direction: inbound (34074 - cpr-j100-0405:51243) [21]
    * transport: TCPROS
 * topic: /tf_static
    * to: /tf_imu (http://cpr-j100-0405:35627/)
    * direction: inbound (39558 - cpr-j100-0405:57601) [22]
    * transport: TCPROS
 * topic: /tf_static
    * to: /tf_gnss (http://cpr-j100-0405:44081/)
    * direction: inbound (50422 - cpr-j100-0405:34431) [29]
    * transport: TCPROS
 * topic: /tf_static
    * to: /tf_vsm (http://cpr-j100-0405:38325/)
    * direction: inbound (59446 - cpr-j100-0405:59081) [30]
    * transport: TCPROS
 * topic: /tf_static
    * to: /tf_aux1 (http://cpr-j100-0405:34213/)
    * direction: inbound (37030 - cpr-j100-0405:50933) [31]
    * transport: TCPROS

```

I know the device is currently reporting nonsense NMEA sentences, but I assumed that the node still would advertise any topics, or am I  wrong there.

Maybe someone can help me with my issue,
thanks in advance.

thomasemter, 2023-07-26:
The driver advertises the topics on first publishing. Since you are using default `rover.yaml`, it should also receive SBF blocks and not just NMEA sentences. According to the output this is not the case.

Can you please provide some more info, like the receiver you are using, how it is connected, ROS and driver version?

Targacore, 2023-07-26:
Thank you for the quick response.

The receiver is a simpleRTK3B Heading. It is connected via USB/Serial (I did not installed it myself, so I can't give you more information on that). The robot runs on ROS melodic and Ubuntu 18.04 and I used the latest driver on the master branch. 
Hope that is enough info.

thomasemter, 2023-07-26:
I had a closer look at the [simpleRTK3B ](https://www.ardusimple.com/simplertk3b-hookup-guide/)and it has two USB ports. Can you please confirm that it is connected to native USB port of the mosaic `Power + GPS`?

Targacore, 2023-07-28:
I have two other tty connections: ``ACM1`` and ``ACM2``. Also tried them as param in the .yaml file with no success. On both I can't receive anything and the parser stops after 10 seconds.

The receiver is build into a robot, so I can not confirm myself is wired up correctly, but I was told everything is connected. I am not able to reach the web interface. Does that mean the native USB connection is missing or do I need some additional driver for that.

thomasemter, 2023-07-31:
I have read further and the names of the virtual serial ports are system dependent. Thus, it is indeed possible they are named `ACMx`. To investigate further, you may try some more things.

- Make sure your username is part of the `dialout` group.
- Connect with a serial terminal like ` screen /dev/ttyACM0` and send `SSSSSSSSSS` (ten times S), the mosaic should answer with something like `USB1>`.
- Install the newest version of the driver via apt or if you have built it from github, use the dev branch. In older versions the initialization of the communication sometimes was unreliable.

The web interface via USB is provided via RNDIS and should be available right after connecting. But I must admit, that I have only  used in Ubuntu 20.04 and 22.04, so I am not 100 % sure it also works in 18.04.

Targacore, 2023-07-31:
I was able to fix the problem.
I connected the receiver directly to my computer and entered the web-interface. There I went to Communication->USB and turned that on.
ACM0 was indeed the wrong interface, it was one of the others.
I'm also using the dev branch now.

Thank you for your help.

closed 2023-07-31 by Targacore

thomasemter, 2023-07-31:
Perfect. Thank you for your feedback.