septentrio-gnss/septentrio_gnss_driver #99: Check confuguration of Ntrip corrections

tiesus, 2023-08-13:
First of all the driver is working pretty well and the documentation helps a lot. The only issue I have, is that I don't get Ntrip connections to work. I use the the Mosaic go heading evaluation kit. <br>
The board is connected via USB port and tcp is used. All the activated messages arrive in ROS. I configured the ntrip settings according to the instructions i found to the following settings:<br> 

    ntrip_1:
     id: "1"
     caster: "www.sapos.geonord-OD.de"
     caster_port: 2101
     username: "gast"
     password: "gast"
     mountpoint: "RTCM4G"
     version: "v2"
     tls: true
     fingerprint: ""
     rtk_standard: "RTCMv3"
     send_gga: "auto"
     keep_open: true

Is there an easy way to check, why ntrip is not working? What i want to check is, if the receiver has internet access and can query corrections from the ntrip caster. Any hints on how to debug this? The web interface of the receiver does not show any activated ntrip streams.

thomasemter, 2023-08-13:
Hi @tiesus ,

the receiver most likely has no internet connection since USB provides a virtual network connection only to your PC.  You have to activate outgoing network access via USB, which is explained [here](https://www.youtube.com/watch?v=bUt8cL9Ue1Y). Additionally, traffic has to be routed between the virtual network port and the interface of your PC connected to the  internet.

tiesus, 2023-08-23:
Hi @thomasemter, 

thanks for your quick response. After setting up the receiver according to your links, everything works fine.

closed 2023-08-23 by tiesus