septentrio-gnss/septentrio_gnss_driver #101: How to connect to device via serial?

tiesus, 2023-09-18:
We need to connect to the device via serial insted aof tcp-Ip, since the the-syscon-driver is not available for arm and we have problems to use the devices on jetson nano. We can connect to the device but did not receive any payload:
`septentrio_gnss: Searching for serial port/dev/ttyACM2
[DEBUG] [1695029925.501838983]: /septentrio_gnss: Leaving initializeIO() method
[DEBUG] [1695029925.501889266]: /septentrio_gnss: Called defineMessages() method
[DEBUG] [1695029925.501941183]: /septentrio_gnss: Key 5914 successfully inserted into multimap: true
[DEBUG] [1695029925.501989459]: /septentrio_gnss: Key $GPGGA successfully inserted into multimap: true
[DEBUG] [1695029925.502032098]: /septentrio_gnss: Key $GPRMC successfully inserted into multimap: true
[DEBUG] [1695029925.502079494]: /septentrio_gnss: Key 4006 successfully inserted into multimap: true
[DEBUG] [1695029925.502143309]: /septentrio_gnss: Key 4007 successfully inserted into multimap: true
[DEBUG] [1695029925.502217710]: /septentrio_gnss: Key 4028 successfully inserted into multimap: true
[DEBUG] [1695029925.502308669]: /septentrio_gnss: Called connect() method
[DEBUG] [1695029925.502367530]: /septentrio_gnss: Started timer for calling reconnect() method until connection succeeds
[DEBUG] [1695029925.502424710]: /septentrio_gnss: Key 5905 successfully inserted into multimap: true
[DEBUG] [1695029925.502506028]: /septentrio_gnss: Key 5906 successfully inserted into multimap: true
[DEBUG] [1695029925.502565937]: /septentrio_gnss: Called reconnect() method
[DEBUG] [1695029925.502633483]: /septentrio_gnss: Key 5908 successfully inserted into multimap: true
[ INFO] [1695029925.502694819]: /septentrio_gnss: Connecting serially to device/dev/ttyACM2, targeted baudrate: 115200
[DEBUG] [1695029925.502727933]: /septentrio_gnss: Calling initializeSerial() method..
[DEBUG] [1695029925.502785033]: /septentrio_gnss: Key 5938 successfully inserted into multimap: true
[DEBUG] [1695029925.502867118]: /septentrio_gnss: Key 5939 successfully inserted into multimap: true
[DEBUG] [1695029925.502941016]: /septentrio_gnss: Key 4027 successfully inserted into multimap: true
[ INFO] [1695029925.502998599]: /septentrio_gnss: Opened serial port /dev/ttyACM2
[DEBUG] [1695029925.503036614]: /septentrio_gnss: Our boost version is 107100.
[DEBUG] [1695029925.503065131]: /septentrio_gnss: Creating new Async-Manager object..
[DEBUG] [1695029925.503136543]: /septentrio_gnss: Key GPST successfully inserted into multimap: true
[DEBUG] [1695029925.503198368]: /septentrio_gnss: Setting the private stream variable of the AsyncManager instance.
[DEBUG] [1695029925.503253391]: /septentrio_gnss: Key NavSatFix successfully inserted into multimap: true
[DEBUG] [1695029925.503334594]: /septentrio_gnss: Key GPSFix successfully inserted into multimap: true
[DEBUG] [1695029925.503414906]: /septentrio_gnss: Key 4013 successfully inserted into multimap: true
[DEBUG] [1695029925.503495424]: /septentrio_gnss: Launching tryParsing() thread..
[DEBUG] [1695029925.503581452]: /septentrio_gnss: Key 4001 successfully inserted into multimap: true
[DEBUG] [1695029925.503652366]: /septentrio_gnss: Called setManager() method
[DEBUG] [1695029925.503692629]: /septentrio_gnss: Leaving setManager() method
[DEBUG] [1695029925.503722766]: /septentrio_gnss: Gradually increasing the baudrate to the desired value...
[DEBUG] [1695029925.503750889]: /septentrio_gnss: Initiated current_baudrate object...
[DEBUG] [1695029925.503787621]: /septentrio_gnss: Key PoseWithCovarianceStamped successfully inserted into multimap: true
[DEBUG] [1695029925.503841758]: /septentrio_gnss: Current baudrate is 115200
[ INFO] [1695029925.503895146]: /septentrio_gnss: Set ASIO baudrate to 115200, leaving InitializeSerial() method
[DEBUG] [1695029925.503977075]: /septentrio_gnss: Key 5902 successfully inserted into multimap: true
[DEBUG] [1695029925.504056902]: /septentrio_gnss: Leaving reconnect() method
[DEBUG] [1695029925.504178742]: /septentrio_gnss: Leaving defineMessages() method
[DEBUG] [1695029925.504256236]: /septentrio_gnss: Called configureRx() method
[DEBUG] [1695029925.504412151]: /septentrio_gnss: Sent the following 22 bytes to the Rx: 
SSSSSSSSSSSSSSSSSSS
[DEBUG] [1695029925.505133951]: /septentrio_gnss: Calling read_callback_() method, with number of bytes to be parsed being 5
[DEBUG] [1695029927.504530274]: /septentrio_gnss: Successully connected. Leaving connect() method
[ INFO] [1695029935.505388750]: /septentrio_gnss: TryParsing() method finished since it did not receive anything to parse for 10 seconds..
`

tiesus, 2023-09-18:
we solved the problem to get payload in the way that we enabled nmea output on the specific port. But the speed is very slow (0,5 hz) and msgs like /gpsfix are never published. The device is an mosaic-H

thomasemter, 2023-09-18:
You are using an older version of the driver, which sometimes had problems to reliably initialize communication. Is it possible for you to use the newest version? Either via apt or if you build from source the `dev` branch in case of ROS 1 or `ros2` branch in case of ROS 2 respectively.

`/gpsfix` needs several SBF blocks including raw measurements, which require quite some bandwidth depending on the desired rate. Is it possible for you to increase the baudrate of the port?

tiesus, 2023-09-18:
We fixed the reading of the messages via serial. We forgot to configure SBF Output in the receiver, so the trigger to publish the message was not available.

closed 2023-09-18 by tiesus

thomasemter, 2023-09-18:
Great. Thank you for your feedback.