septentrio-gnss/septentrio_gnss_driver #102: No messages received via usb

tiesus, 2023-09-20:
I have a problem connecting to the Mosaic-X5  Board via USB connection. The setup is complete but the data streams never start: 
Driver output, where the receiver hangs: `[ INFO] [1695217621.574985984]: /septentrio_gnss: Setup complete.
[DEBUG] [1695217621.575028507]: /septentrio_gnss: Successully connected. Leaving connect() method
[DEBUG] [1695217621.575107190]: /septentrio_gnss: Leaving ROSaicNode() constructor..
[DEBUG] [1695217621.575844638]: /septentrio_gnss: A message received:   SBFOutput, Stream3, USB2, MeasEpoch+AttEuler+PVTCartesian+PVTGeodetic+PosCovGeodetic+VelCovGeodetic+DOP+ReceiverTime+ChannelStatus+AttCovEuler, msec50

[DEBUG] [1695217621.575914777]: /septentrio_gnss: handleCd: USB2>`

[image]

thomasemter, 2023-09-20:
Can you please share your config and what version of the driver you are using?

tiesus, 2023-09-20:
Config gnss.yaml: 

```# Configuration Settings for the Rover Rx

# GNSS/INS Parameters

# We set this with a udev rule. Use /dev/ttyACM0 or so in if you don't have udev rule set
device: serial:/dev/septentrioMosaic

serial:
  baudrate: 921600
  hw_flow_control: "off"

stream_device:
  tcp:
    ip_server: ""
    port: 0
  udp:
    ip_server: ""
    port: 0
    unicast_ip: ""

configure_rx: true

login:
  user: ""
  password: ""

osnma:
  mode: "loose"
  ntp_server: ""
  keep_open: true

frame_id: gps

aux1_frame_id: gps_base

get_spatial_config_from_tf: true

use_ros_axis_orientation: true

receiver_type: gnss

multi_antenna: true

datum: Default

att_offset:
  heading: 90.0
  pitch: 0.0

ant_type: "Unknown"
ant_serial_nr: "Unknown"
ant_aux1_type: "Unknown"
ant_aux1_serial_nr: "Unknown"

polling_period:
  pvt: 50
  rest: 0

use_gnss_time: false

latency_compensation: false

rtk_settings:  
  ntrip_1:
    id: ""
    caster: ""
    caster_port: 2101
    username: ""
    password: ""
    mountpoint: ""
    version: "v2"
    tls: false
    fingerprint: ""
    rtk_standard: "auto"
    send_gga: "auto"
    keep_open: true
  ip_server_1:
    id: ""
    port: 0
    rtk_standard: "auto"
    send_gga: "auto"
    keep_open: true
  serial_1:
    port: ""
    baud_rate: 115200
    rtk_standard: "auto"
    send_gga: "auto"
    keep_open: true

publish:
  # For both GNSS and INS Rxs
  navsatfix: true
  gpsfix: true
  gpgga: true
  gprmc: true
  gpst: true
  measepoch: true
  pvtcartesian: true
  pvtgeodetic: true
  basevectorcart: false
  basevectorgeod: false
  poscovcartesian: true
  poscovgeodetic: true
  velcovcartesian: false
  velcovgeodetic: true
  atteuler: true
  attcoveuler: true
  pose: true
  twist: true
  diagnostics: true
  aimplusstatus: true
  galauthstatus: false
  # For GNSS Rx only
  gpgsa: false
  gpgsv: false

# logger

activate_debug_log: true```

The driver version is the version from the dev branch.

thomasemter, 2023-09-20:
Thank you for the config. I tried it with both USB serial interfaces and it worked on my setup.

I am sorry, but I am a little clueless right now why it dos not work in your setup. Does your udev rule map `/dev/ttyACM0` to `/dev/septentrioMosaic` ? Because on my machine `/dev/ttyACM0` has the connection descriptor `USB1` and `/dev/ttyACM1` has `USB2`. Your debug output shows `USB2`.

tiesus, 2023-09-20:
I think the Udev rule is causing the error. I created a symlink to `/dev/septentrioMosaic`. I managed now to make everything work on the jetson nano with `tcp` connection without internet sharing. Thanks for your help:)

closed 2023-09-20 by tiesus

thomasemter, 2023-09-21:
Great you could work it out.

Can you please share what was the trick to get it work on Jetson with `tcp`?

thomasemter, 2023-09-22:
Ok, thank you for the explanation.

I indeed also had a look at this ntrip_client some time ago but since we have our receivers connected via real Ethernet we did not have a need for it so far.

The option to publish a generic NMEA sentence is on my TODO list but I do not have much time right now.