septentrio-gnss/septentrio_gnss_driver #104: Angular velocity units in topic /imu

abaeyens-imod, 2023-10-18:
### Issue description
The angular velocities published in the topic `/imu` appear to be expressed in unit deg/s instead of rad/s. Following the ROS message definition, it was expected that they are expressed in the latter and not the former units.

As a result, all existing ROS nodes (which assume rad/s) get the impression that the platform is rotating about 60 times faster than in reality, which results in control issues.

### Steps to reproduce the issue
1. Setup Septentrio hardware and launch Septentrio ROS node.
2. Log output of topic `/imu`.
3. Rotate Septentrio setup at a known angular velocity, eg. 15°/s.
4. Read/visualized the recorded angular velocities in log file.
5. Compare the amplitude of the logged values with the applied known velocity.

### What's the expected result?
The angular velocities are expressed in unit rad/s
- as defined in the [ROS message definition of the message `sensor_msgs/msg/Imu`](https://docs.ros2.org/foxy/api/sensor_msgs/msg/Imu.html) , the type of the message published on the `/imu` topic.
- and in line with [REP 103](https://www.ros.org/reps/rep-0103.html#units), which recommends SI units.

### What's the actual result?
The values are about 60 times larger than expected. They are probably expressed in °/s instead of rad/s

### Additional details / screenshot
Graph created when rotating the Septentrio at a sine-modulated angular velocity in the order of 15 °/s:
[image]
*Visualization using Plotjuggler*

[image]
*Visualization using rqt*

### Version
Compiled `septentro_gnss_driver` from source, branch `ros2`, HEAD commit `207c37bc`.

We would appreciate if you could look into this. In case any other information is desired, please let know.

thomasemter, 2023-10-18:
Thank you for the detailed bug report. This is indeed handled incorrect and will be fixed.

thomasemter, 2023-10-18:
For a quick fix, you may change the following 3 lines: https://github.com/thomasemter/septentrio_gnss_driver/blob/ece8d66758843a8f95bf615894eb6374bd181e53/src/septentrio_gnss_driver/communication/message_handler.cpp#L558

abaeyens-imod, 2024-02-26:
Thank you for resolving this!

closed 2024-02-26 by abaeyens-imod