septentrio-gnss/septentrio_gnss_driver #107: Namespace handling

abaeyens-imod, 2023-10-19:
### Issue description
Specifying a namespace when launching the Septentrio node does not effect the names of the published topic - they are always published in the root name space.

For topics like `/insnavgeod` this is not much of an issue - it's clear they're from the Septentrio and there aren't other nodes that publish such topics - but for others like `/imu` and `/pose` this will cause confusion.

### Steps to reproduce the issue
1. Setup Septentrio hard & software
2. Specify which namespace to launch the Septentrio nodes with in the Septentrio launch file
3. Launch with that launch file
4. Look at the names of the published topics

### What's the expected result?
```bash
arne@laptop:~$ ros2 topic list
/<namespace>/insnavgeod
/<namespace>/imu
/<namespace>/pose
etc.
```

### What's the actual result?
```bash
arne@laptop:~$ ros2 topic list
/insnavgeod
/imu
/pose
etc.
```

### Additional details
This is likely due to the topics being defined with a preceding slash in `septentrio_gnss_driver/src/septentrio_gnss_driver/communication/message_handler.cpp`, e.g.
```cpp
publish<INSNavCartMsg>("/insnavcart", last_insnavcart_);
```
instead of
```cpp
publish<INSNavCartMsg>("insnavcart", last_insnavcart_);
```

### Version### Issue description
Specifying a namespace when launching the Septentrio node does not effect the names of the published topic, they are always published in the root name space.

For topics like `/insnavgeod` this is not much of an issue, it's clear they're from the Septentrio and there aren't other nodes that publish such topics, but for others like `/imu` and `/pose` this will cause confusion.

### Steps to reproduce the issue
1. Setup Septentrio hard & software
2. Specify which namespace to launch the Septentrio nodes with in the Septentrio launch file
3. Launch with that launch file
4. Look at the names of the published topics

### What's the expected result?
```bash
arne@laptop:~$ ros2 topic list
/<namespace>/insnavgeod
/<namespace>/imu
/<namespace>/pose
etc.
```

### What's the actual result?
```bash
arne@laptop:~$ ros2 topic list
/insnavgeod
/imu
/pose
etc.
```

### Additional details
This is likely due to the topics being defined with a preceding slash in `septentrio_gnss_driver/src/septentrio_gnss_driver/communication/message_handler.cpp`, e.g.
```cpp
publish<INSNavCartMsg>("/insnavcart", last_insnavcart_);
```
instead of
```cpp
publish<INSNavCartMsg>("insnavcart", last_insnavcart_);
```

### Version
Compiled `septentro_gnss_driver` from source, branch `ros2`, HEAD commit `207c37bc`.

Thank you in advance for your time. In case there's any setting we missed, or more information is required, please share.
Compiled `septentro_gnss_driver` from source, branch `ros2`, HEAD commit `207c37bc`.

Thank you in advance for addressing this. Should you need more details or information, please don't hesitate to ask.

thomasemter, 2023-10-19:
Thank you for the report. This is indeed handled incorrectly and will be fixed.

abaeyens-imod, 2024-02-26:
Thank you for resolving this!

closed 2024-02-26 by abaeyens-imod