septentrio-gnss/septentrio_gnss_driver #112: Invalid or erroneous data like -20000000 in navsatfix messages

ayoubasritii, 2023-11-23:
Hello,

I am using the Septentrio INS SBI Pro + and the official driver for my project, I have encountered an issue where the `navsatfix` topic occasionally publishes invalid values, specifically `-20000000`, under certain conditions (e.g., bad GNSS signal). This behavior can lead to complications in downstream applications such as localization and navigation systems.

**Description of the issue:**
- The `navsatfix` messages from the driver sometimes contain latitude and longitude values of `-20000000`.
- The status field of the `navsatfix` message during these instances is -1.

**Impact:**
- These invalid values can disrupt the performance of localization systems that rely on accurate GPS data.
- Filtering these values externally is a workaround but not an ideal solution, as it adds complexity and potential points of failure in the system.

**Expected behavior:**
- It would be preferable for the driver to not publish `navsatfix` messages under these conditions.

**Environment:**
- ROS Version: noetic
- Septentrio Driver Version: 1.3.2 (2023-11-19)
- Hardware/Platform: Ubuntu 20.04

I am looking forward to any suggestions, guidance, or fixes you can provide. Additionally, if there's any further information or testing I can assist with, please let me know.

Best regards,
Ayoub.

thomasemter, 2023-11-23:
Hello Ayoub,

the driver works as intended and correctly flags the fix status as `STATUS_NO_FIX`. This behavior allows for the system to distinguish between a communication failure from the receiver and invalid data.

In order to process the data, it is a must to always evaluate the status flag. Nonetheless, we are contemplating about changing invalid data to NaN to make it more obvious.

Best regards,
Thomas

ayoubasritii, 2023-11-23:
Hello Thomas,
Thank you for your prompt reply.
I agree that the status flag should be evaluated, but according to this [workshop](https://www.youtube.com/watch?v=PFSxcOPnfjQ&t=2s&ab_channel=Septentrio) the driver should be able to be used by the robot localization package. As far as I know, this package does not handle invalid or NaN data or am I mistaken?

thomasemter, 2023-11-23:
robot_localization does check for [good_gps](https://github.com/cra-ros-pkg/robot_localization/blob/804729e7bf7d1b83e2dc4b33fc8ff9cf4f2dd9a2/src/navsat_transform.cpp#L614), i.e., for both valid status and not NaN.

closed 2023-12-12 by ayoubasritii