septentrio-gnss/septentrio_gnss_driver #113: ntrip of rtk_settings does not work for ROS 1 (ROS Noetic) driver on Ubuntu 20.04

hangcui1201, 2024-02-27:
Hello,

We are using AsteRx SBi3 Pro+ for a self-driving project. We are using ROS Noetic (ROS 1) on Ubuntu 20.04. **We connected the AsteRx SBi3 Pro+ through the USB cable and the computer had Internet access.** **We have the RTK service from SmartNet**. 

However, when we did the settings as below, the RTK **did not work.** 

rtk_settings:
  keep_open: true
  ntrip_1:
    id: ""
    caster: "98.159.149.72"
    caster_port: 9301
    username: "uni01317601"
    password: "xxxx"
    mountpoint: "MSM_iMAX"
    version: "v2"
    tls: false
    fingerprint: ""
    rtk_standard: "auto"
    send_gga: "auto"

We understand that on Windows we have to share the Internet over USB. However, we are using the Ubuntu linux. We are **not sure** whether the AsteRx SBi3 Pro+ has Internet access or not when it is connected to the computer through the USB cable.

**We want to make the RTK work together with ROS. Could let us know how to configure the AsteRx SBi3 Pro+ device?**

Thanks!

Best,
Hang

thomasemter, 2024-02-28:
Hello Hang,

you observed correctly, that you need internet forwarding over USB. How to achieve this is explained [here](https://github.com/septentrio-gnss/septentrio_gnss_driver/issues/99#issuecomment-1676480102).

Best regards,
Thomas

hangcui1201, 2024-02-29:
Hi Thomas,

Thanks a lot. I have made it work. A quick question. **Could you tell me know in which ROS topic I could verify that the RTK is working?** For example, rtk: true, or GNSS mode: rtk, etc. 

[image]

[image]

Thanks again. 

Best,
Hang

thomasemter, 2024-02-29:
Hi Hang,

great :-)

The `mode` field of `pvt_geodetic` or `pvt_cartesian` shows the type of the PVT solution:
[image]

Cheers,
Thomas

hangcui1201, 2024-03-03:
Hi Thomas,

Thanks! I could tell the RTK mode from the ROS topic now. 

Another question. Can I use the Ethernet interface instead of the USB interface for above settings? My device is AsteRx SBi3 Pro+. The reason is that I found the USB interface of my computer could not provide stable power to the device. 

**I want to power the AsteRx SBi3 Pro+ separately and use the Ethernet interface for ROS and RTK settings. Could let me how to configure it? Can I change the IP address of AsteRx SBi3 Pro+?**

Thanks a lot!

Best,
Hang

thomasemter, 2024-03-04:
Hi Hang,

you can use Ethernet for sure. I would recommend to connect it to a switch then, so you won't need internet sharing from your PC anymore.

The IP address can be changed via web interface Communication -> Ethernet. For ROSaic you just have to change the IP address.

Cheery,
Thomas

hangcui1201, 2024-03-04:
Hi Thomas,

Thanks a lot! I will try it. 

I find the IMU outputs from ROS topic /septentrio_gnss/imu include the gravity compensation. Is there anyway to get the **raw data of the IMU without gravity compensation**?

Thanks again.

Best,
Hang

thomasemter, 2024-03-08:
Hi Hang,

are you really sure about the gravity compensation? The raw data of the IMU shows the gravity:
[image]

Cheers,
Thomas

hangcui1201, 2024-03-08:
Hi Thomas,

Yes, in the linear_acceleration. In the Z direction, it added the gravity component, which is 9.8 g/m^s. **Can we have the IMU raw data without the gravity components?**

Thanks!

Best,
Hang

thomasemter, 2024-03-08:
Hi Hang,

to be clear: the IMU measures gravity, so raw data includes gravity of course. Else, it would not be possible to determine the attitude ;-)

In order to get gravity compensated accelerations, gravity has to be subtracted. This is not straight forward, since it is dependent on various factors. Maybe we can add this as an option to a future version of the driver.

Cheers,
Thomas