septentrio-gnss/septentrio_gnss_driver #114: Mosaic X5: Ntrip of rtk_settings does not work for ROS 1 (ROS Noetic) driver on Ubuntu 20.04

echoGee, 2024-02-27:
We are using the following setting file on the yaml file. The ntrip settings work using the datalink, but not the ros driver. 

```
# Configuration Settings for the Rover Rx

# GNSS/INS Parameters

device: tcp://192.168.3.1:28784

serial:
  baudrate: 921600
  hw_flow_control: "off"

stream_device:
  tcp:
    ip_server: ""
    port: 0
  udp:
    ip_server: ""
    port: 0
    unicast_ip: ""

configure_rx: true

login:
  user: ""
  password: ""

osnma:
  mode: "loose"
  ntp_server: ""
  keep_open: true

frame_id: gnss

aux1_frame_id: aux1

get_spatial_config_from_tf: true

use_ros_axis_orientation: true

receiver_type: gnss

multi_antenna: false

datum: Default

att_offset:
  heading: 0.0
  pitch: 0.0

ant_type: "Unknown"
ant_serial_nr: "Unknown"
ant_aux1_type: "Unknown"
ant_aux1_serial_nr: "Unknown"

polling_period:
  pvt: 0
  rest: 500

use_gnss_time: false

latency_compensation: false

rtk_settings:  
  ntrip_1:
    id: "RTS1"
    caster: "xx.com"
    caster_port: 2101
    username: "aa"
[... 330 lines cut ...]
[DEBUG] [1709075596.035569626]: /septentrio_gnss: A message received:   NMEATalkerID, GP

[DEBUG] [1709075596.035582183]: /septentrio_gnss: handleCd: IP10>
[DEBUG] [1709075596.035636264]: /septentrio_gnss: AsyncManager sent the following 40 bytes to the Rx: sno, Stream2, IP10, +GGA +RMC, OnChange
[DEBUG] [1709075596.108394140]: Trying to publish message of type [nmea_msgs/Gprmc/02533bac67f17457b2e3538525ba1aae] on a publisher with type [nmea_msgs/Gprmc/02533bac67f17457b2e3538525ba1aae]
[DEBUG] [1709075596.109505008]: Trying to publish message of type [nmea_msgs/Gpgga/8f51cb504898f39d8ad9f698f0b6f9cd] on a publisher with type [nmea_msgs/Gpgga/8f51cb504898f39d8ad9f698f0b6f9cd]
[DEBUG] [1709075596.126800061]: /septentrio_gnss: The Rx's response contains 45 bytes and reads:
 $R: sno, Stream2, IP10, +GGA +RMC, OnChange

[DEBUG] [1709075596.126878306]: /septentrio_gnss: A message received:   NMEAOutput, Stream2, IP10, GGA+RMC, OnChange

[DEBUG] [1709075596.126915276]: /septentrio_gnss: handleCd: IP10>
[DEBUG] [1709075596.127053735]: /septentrio_gnss: AsyncManager sent the following 174 bytes to the Rx: sso, Stream3, IP10, +ReceiverTime +PVTCartesian +PVTGeodetic +PosCovCartesian +PosCovGeodetic +VelCovGeodetic +AttEuler +AttCovEuler +MeasEpoch +ChannelStatus +DOP, OnChange
[DEBUG] [1709075596.188288429]: /septentrio_gnss: The Rx's response contains 179 bytes and reads:
 $R: sso, Stream3, IP10, +ReceiverTime +PVTCartesian +PVTGeodetic +PosCovCartesian +PosCovGeodetic +VelCovGeodetic +AttEuler +AttCovEuler +MeasEpoch +ChannelStatus +DOP, OnChange

[DEBUG] [1709075596.188505461]: /septentrio_gnss: A message received:   SBFOutput, Stream3, IP10, MeasEpoch+AttEuler+PVTCartesian+PosCovCartesian+PVTGeodetic+PosCovGeodetic+VelCovGeodetic+DOP+ReceiverTime+ChannelStatus+AttCovEuler, OnChange

[DEBUG] [1709075596.188548986]: /septentrio_gnss: handleCd: IP10>
[DEBUG] [1709075596.188709547]: /septentrio_gnss: Leaving configureRx() method
[ INFO] [1709075596.188769319]: /septentrio_gnss: Setup complete.
[DEBUG] [1709075596.188799802]: /septentrio_gnss: Successully connected. Leaving connect() method
[DEBUG] [1709075596.188869958]: /septentrio_gnss: Leaving ROSaicNode() constructor..

```

thomasemter, 2024-02-28:
Hi @echoGee ,

did you activate internet forwarding? An explanation on how to set this up can be found [here[(https://github.com/septentrio-gnss/septentrio_gnss_driver/issues/99#issuecomment-1676480102).

Best regards,
Thomas

echoGee, 2024-02-28:
Will try this. Is this needed even when using the serial communication instead of tcp ? 

> Hi @echoGee ,
> 
> did you activate internet forwarding? An explanation on how to set this up can be found [here](https://github.com/septentrio-gnss/septentrio_gnss_driver/issues/99#issuecomment-1676480102).
> 
> Best regards, Thomas

thomasemter, 2024-02-28:
It is needed since serial does not support NTRIP. Serial supports RTCM directly instead.