septentrio-gnss/septentrio_gnss_driver #117: Septentrio AsteRx SBi3 Pro+ RTK status jumps and GNSS/INS data invalid with ROS driver

hangcui1201, 2024-04-01:
Hi,

We are using the septentrio_gnss_driver v1.3.2 (https://github.com/septentrio-gnss/septentrio_gnss_driver/releases/tag/v1.3.2) for AsteRx SBi3 Pro+ on Ubuntu 2.04 with ROS Noetic. We used the USB interface with AsteRx SBi3 Pro+. 

**We had two issues related to RTK and GNSS/INS output data**. Before we use the device, I did the device initialization from this link (https://customersupport.septentrio.com/s/article/How-to-initialize-the-IMU-of-your-Septentrio-GNSS-INS-solution). 

(1) We used HxGN SmartNet RTK service. We had the RTK Fixed worked. **However, we found that whenever we drove our vehicle around, the RTK fixed status jumps to Disabled and the GNSS/INS output data also drifted**

Note: I understand that which ROS topic to read for the status of the GNSS/INS, such as RTK float, RTK fixed, or no RTK in previous post. We checked the Internet connect, it was good for the 4G/LTE module. 

[image]

[image]

(2) We also found that **the GNSS/INS data became invalid** quite often when we drove the vehicle outside. Sometime the invalid data stayed for quite a long time or never getting back to the normal values. 

[image]

We also posted our recorded video here: https://youtu.be/jCihYPSxACE

We configured the AsteRx SBi3 Pro+ device through the webGUI, not the ROS driver as belows. 

[image]

[image]

[image]

**Any suggestion on above issues?**

Thanks a lot for the help!

Best,
Hang

hangcui1201, 2024-04-03:
Hi,

Following the previous post, we also found the Main and Aux1 signals were pretty bad. However, Main & Aux1 RF power were good. We used TW7972 triple band GNSS antennas. **Any suggestions?**

[image]

Thanks! 

Best,
Hang

thomasemter, 2024-04-03:
Hi Hang,

this does not seem to be driver related and I believe [Septentrio's support](https://customersupport.septentrio.com/s/) is more proficient to help in this case.

Best regards,
Thomas

hangcui1201, 2024-04-03:
Hi Thomas,

Thanks a lot! I will submit a problem to the Septentrio's support.

Best,
Hang