septentrio-gnss/septentrio_gnss_driver #119: Unable to enable NTRIP via ROS yaml

fiorano10, 2024-04-04:
I am trying to enable and set up ntrip via the ROS driver but it does work. If I do it via the web interface it works fine but I am unable to change the params and have them take effect. I get the following error when running the node.

```
[ INFO] [1712269543.763666999]: /mosaic_rover: Setting up Rx.
[ERROR] [1712269544.007263301]: /mosaic_rover: Invalid command just sent to the Rx! The Rx's response contains 62 bytes and reads:
 $R? setAntennaOffset: Tabular argument 'Antenna' is invalid!

[ERROR] [1712269544.008559705]: /mosaic_rover: Invalid command just sent to the Rx! The Rx's response contains 57 bytes and reads:
 $R? setNtripSettings: Tabular argument 'Cd' is invalid!

[ERROR] [1712269544.010540076]: /mosaic_rover: Invalid command just sent to the Rx! The Rx's response contains 57 bytes and reads:
 $R? setNtripSettings: Tabular argument 'Cd' is invalid!

[ERROR] [1712269544.011905323]: /mosaic_rover: Invalid command just sent to the Rx! The Rx's response contains 60 bytes and reads:
 $R? setNtripTlsSettings: Tabular argument 'Cd' is invalid!
```

RTK settings in the yaml
```

rtk_settings:
  ntrip_1:
    id: "NTRIP1"
    caster: ""
    caster_port: 2101
    username: ""
    password: ""
    mountpoint: ""
    version: "v2"
    tls: false
    fingerprint: ""
    rtk_standard: "auto"
    send_gga: "auto"
    keep_open: true
  ip_server_1:
    id: ""
    port: 0
    rtk_standard: "auto"
    send_gga: "auto"
    keep_open: true
  serial_1:
    port: ""
    baud_rate: 115200
    rtk_standard: "auto"
    send_gga: "auto"
    keep_open: true

```

fiorano10, 2024-04-09:
Yup - that was it. Thank you!

thomasemter, 2024-04-10:
Great, thanks for confirming,