septentrio-gnss/septentrio_gnss_driver #120: SNR Logging

coen132, 2024-04-10:
Hello,

Is there a quick way to record the Signal Noise Ratio on a ROS2 topic? I saw the C/n0 is used internally with the MeasEpochChannelType1(2).msg but it is not published to a topic.

Thanks in advance!

thomasemter, 2024-04-10:
Hello,

MeasEpochChannelType1/2 are part of the MeasEpoch message and published with the parameter `publish.measepoch` set to `true`. Also, the SNRs are part of the GPSFix message.

Cheers!

closed 2024-04-10 by coen132

coen132, 2024-04-11:
Thanks for your response, the MeasEpoch I figured out but I can't find the SNR's in the GPS fix msg, I'm not sure but I think this: http://docs.ros.org/en/hydro/api/gps_common/html/msg/GPSFix.html  is used?

Thanks, Coen

thomasemter, 2024-04-11:
Hi Coen,

The GPSFix message has a field `status` of type https://docs.ros.org/en/hydro/api/gps_common/html/msg/GPSStatus.html where you can find the SNRs.

coenheerinkgravis, 2024-05-16:
@thomasemter I recorded some SNR data but is there a table/manual with ranges of acceptable SNR values?

thomasemter, 2024-05-16:
I am sorry, that is a question to which [Septentrio's support](https://www.septentrio.com/en/support) can give you a more informed answer.