septentrio-gnss/septentrio_gnss_driver #121: integration of /localization odomery data to robot_localization node

NK1807, 2024-04-28:
Hello 
I have been using  septentrio's Asterx SBi3 pro+ with dual antenna setup. I have integrated it to robot_localization node via navsat_transform .
I would like to integrate directly /localization odometry topic data as it gives absolute heading. My doubts/questions are below
1) The frame ids of the /localization is "utm_32n" to child frame "base_link". How can i integrate this to robot _localization node as it needs map -> base_link? (to second ekf_map instance as  I'm integrating lidar odom and IMU from INS receiver to first ekf instance)  
2)which is the better way to integrate. is it navsatfix to navsat_transform or /localization to ekf_map instance? 
3)this is a doubt, does /imu from this receiver already fused with dual antenna heading information?

Could some please clarify these?

working environment:
ROS2 umble
Jetson Orin AGX - ubuntu 22

thomasemter, 2024-04-29:
Hello,

I am not 100 % familiar with robot_localization but I will try to answer your questions to the best of my knowledge.

1. Since UTM is a map projection it actually is a map frame. If you have a different map frame you may provide a static transform between the two. Or you may provide UTM coordinates directly in the yaml file of your map.
2. I am not really sure what ekf_map represents but `localization` provides the pose in a Cartesian frame whereas `navsatfix` provides geodetic coordinates.
3. Orientation contains dual antenna heading information as long as there is proper GNSS reception.

I hope this helps.

NK1807, 2024-04-29:
Thanks for the clarification @thomasemter