septentrio-gnss/septentrio_gnss_driver #122: /imu give .nan in orientation

NK1807, 2024-05-09:
Hello
I'm using ros2 (branch ros2) driver for AsteRx SBi3 Pro+ on Ubuntu 20.04 with ROS Foxy.  I have been receiving ".nan" in the orientation field of the /imu topic. Everything else seems good. I have received the RTK FIxed status and connected to base station.
I also tried to checkout to dev2 branch but couldn't able to build it on foxy due to some errors.
Note* I also used drivers from dev2 branch with ros2 humble. it was built and working fine. /imu topic is also having accurate values.

Could someone help me out with this issue?

working environment:
ROS Foxy
Jetson Orin Nano

thomasemter, 2024-05-09:
Hello,
there was indeed a bug which lead to the orientation of the IMU message being NaN. This has been fixed in the dev2 branch. Unfortunately, an incompatibility with Foxy has been introduced, which has gone unnoticed. Thank you for discovering this. I have fixed in my [develpoement repo](https://github.com/thomasemter/septentrio_gnss_driver/tree/devel). Please try it and report back if there are still problems.

NK1807, 2024-05-10:
Your repo is working fine @thomasemter . Thanks for the prompt support. :)

thomasemter, 2024-05-12:
Great, Thanks for your feedback.