septentrio-gnss/septentrio_gnss_driver #128: -20000000000 orientation values in dual-antenna setup

remimotio, 2024-05-30:
Good afternoon,

We are using the mosaic-H module in dual-antenna setup and have checked that the "multi-antenna" mode was selected, both in the config/rover.yaml that we are using and in the web interface. 

Nonetheless, we are still faced with a value of -20000000000 in the fields "track", "pitch" and "roll" as long as the mosaic-H module is not moving, and as soon as it's not moving anymore. These fields do provide us with correct values when moving, but it's as if the module were in "MovingBase" mode (which it's not supposed to be in). 

Moreover, contrary to another issue we read, in the "/atteuler" topic the "heading" field also displays a value of -20000000000 when the mosaic-H is at a standstill.

Could the issue stem from septentrio_gnss_driver?

Thank you very much for your time and attention

thomasemter, 2024-05-30:
Hi,

both issues are not due to the driver.

I assume you mean the fields of `GPSFix`, right? Track is not the same as heading and can indeed only be measured in motion. Think of a drone flying sideways, where the direction of motion (track) does not coincide with the direction it is heading. To be able to submit heading in the `GPSFix` message, we misuse the field `dip` for heading, see [issue 78](https://github.com/septentrio-gnss/septentrio_gnss_driver/issues/78).

If `heading` is not valid in `atteuler`, the mosaic module could not compute a heading. This might be because of too few common satellites being seen by both antennas. One reason for this could be an obstructing structure between the antennas. For these kind of issues [Septentrio's support](https://www.septentrio.com/en/support) is better able to help.

remimotio, 2024-05-31:
Thank you very much for this quick answer!

We'll run some tests based on your suggestions, and hopefully come back to close this issue.

Have a nice day!