septentrio-gnss/septentrio_gnss_driver #129: Serial ID of mosiac go heading not constant

rajsurya1012, 2024-06-06:
Hey,
I have a MOSIAC GO heading evaluation kit. When searching for the device using "serial/by-id", I have two names for one device. Only when the connection is established using one of the ids, I am able to receive ROS messages. When using the other id, connection is established but there are no messages.
Moreover,  the id through which the messages are received is not constant and changes between the two upon restart.

How do I fix it and why is it happening?

Thank you!

thomasemter, 2024-06-07:
Hi,

I am sorry to hear that since I am not really sure what is going on. I always use the RNDIS connection, so I am not so familiar with the serial connections. There are indeed two serial devices by design - `USB1` and `USB2`, see manual. On my system assignments to the two `/dev/serial/by-id` are always constant. 

Regarding message reception, I have tried both and have the same experience as you: only on `USB2` message reception is possible. But this is also constant.

Maybe you could also contact [Septentrio's support](https://www.septentrio.com/en/support) for assistance since this seems not to be related to the ROS driver.

thomasemter, 2024-06-07:
With `USB1` the streams do show up in the web UI correctly. Strangely enough, after reactivating them manually in the web UI, setup via the driver worked afterwards on both connections.

rajsurya1012, 2024-06-07:
ok, thank you!

closed 2024-06-07 by rajsurya1012