septentrio-gnss/septentrio_gnss_driver #132: Asterx-SBI 3 pro INS: discrepancy between /navsatfix.status.status and /gpsfix.status.status with NMEA msgs recorded from

federicobenna, 2024-07-17:
I have mobile robot setup with INS receiver + dual antenna setup and an active subscription with a local RTK provider over internet.

I have been observing a discrepancy between reported status/ gps quality from ros msgs ` /gpsfix.status.status` ` /navsatfix.status.status`  and what the Septentrio web ui and the actual NMEA msg logged on the receiver itself.

[image]

On my tests I see msgs published with status 0 most of the time and jumps to 2 (augmented fix) a couple of times over multiple experiments while the frontend shows Fixed pose and GGA NMEA shows quality indicator 4 (`RTK Fixed`) and GNS mode indicator RRRN (`Real-Time Kinematic. Satellite system used in RTK mode with fixed integers`):

```
<google-sheets-html-origin><style type="text/css"><!--td {border: 1px solid #cccccc;}br {mso-data-placement:same-cell;}--></style>
$INGNS | 90200 | 5051.890218 | N | 440.1211738 | E | RRRN | 13 | 0.9
$INGGA | 90200.5 | 5051.890219 | N | 440.1211744 | E | 4 | 15 | 0.8
$INGNS | 90200.5 | 5051.890219 | N | 440.1211744 | E | RRRN | 15 | 0.8
$INGGA | 90201 | 5051.890217 | N | 440.121174 | E | 4 | 14 | 0.8
$INGNS | 90201 | 5051.890217 | N | 440.121174 | E | RRRN | 14 | 0.8
$INGGA | 90201.5 | 5051.890216 | N | 440.121175 | E | 4 | 15 | 0.8
$INGNS | 90201.5 | 5051.890216 | N | 440.121175 | E | RRRN | 15 | 0.8
$INGGA | 90202 | 5051.890216 | N | 440.1211736 | E | 4 | 15 | 0.8
$INGNS | 90202 | 5051.890216 | N | 440.1211736 | E | RRRN | 15 | 0.8
$INGGA | 90202.5 | 5051.890216 | N | 440.1211737 | E | 4 | 15 | 0.8
$INGNS | 90202.5 | 5051.890216 | N | 440.1211737 | E | RRRN | 15 | 0.8
$INGGA | 90203 | 5051.890216 | N | 440.1211733 | E | 4 | 15 | 0.8
$INGNS | 90203 | 5051.890216 | N | 440.1211733 | E | RRRN | 15 | 0.8
$INGGA | 90203.5 | 5051.890215 | N | 440.1211735 | E | 4 | 15 | 0.8
$INGNS | 90203.5 | 5051.890215 | N | 440.1211735 | E | RRRN | 15 | 0.8
$INGGA | 90204 | 5051.890218 | N | 440.1211742 | E | 4 | 15 | 0.8
$INGNS | 90204 | 5051.890218 | N | 440.1211742 | E | RRRN | 15 | 0.8
$INGGA | 90204.5 | 5051.890218 | N | 440.1211734 | E | 4 | 15 | 0.8
$INGNS | 90204.5 | 5051.890218 | N | 440.1211734 | E | RRRN | 15 | 0.8
$INGGA | 90205 | 5051.890218 | N | 440.1211725 | E | 4 | 15 | 0.8
$INGNS | 90205 | 5051.890218 | N | 440.1211725 | E | RRRN | 15 | 0.8
$INGGA | 90205.5 | 5051.890218 | N | 440.1211722 | E | 4 | 15 | 0.8
$INGNS | 90205.5 | 5051.890218 | N | 440.1211722 | E | RRRN | 15 | 0.8
$INGGA | 90206 | 5051.890218 | N | 440.1211727 | E | 4 | 15 | 0.8
$INGNS | 90206 | 5051.890218 | N | 440.1211727 | E | RRRN | 15 | 0.8
$INGGA | 90206.5 | 5051.890217 | N | 440.1211729 | E | 4 | 15 | 0.8
$INGNS | 90206.5 | 5051.890217 | N | 440.1211729 | E | RRRN | 15 | 0.8
$INGGA | 90207 | 5051.890217 | N | 440.1211733 | E | 4 | 15 | 0.8
$INGNS | 90207 | 5051.890217 | N | 440.1211733 | E | RRRN | 15 | 0.8
$INGGA | 90207.5 | 5051.890217 | N | 440.1211732 | E | 4 | 15 | 0.8
$INGNS | 90207.5 | 5051.890217 | N | 440.1211732 | E | RRRN | 15 | 0.8
$INGGA | 90208 | 5051.890217 | N | 440.1211726 | E | 4 | 15 | 0.8
$INGNS | 90208 | 5051.890217 | N | 440.1211726 | E | RRRN | 15 | 0.8
$INGGA | 90208.5 | 5051.890217 | N | 440.1211724 | E | 4 | 15 | 0.8
$INGNS | 90208.5 | 5051.890217 | N | 440.1211724 | E | RRRN | 15 | 0.8
$INGGA | 90209 | 5051.890216 | N | 440.1211721 | E | 4 | 15 | 0.8
$INGNS | 90209 | 5051.890216 | N | 440.1211721 | E | RRRN | 15 | 0.8
$INGGA | 90213.5 | 5051.890218 | N | 440.1211728 | E | 4 | 15 | 0.8
$INGNS | 90213.5 | 5051.890218 | N | 440.1211728 | E | RRRN | 15 | 0.8
$INGGA | 90214 | 5051.890218 | N | 440.1211725 | E | 4 | 15 | 0.8
$INGNS | 90214 | 5051.890218 | N | 440.1211725 | E | RRRN | 15 | 0.8
$INGGA | 90214.5 | 5051.890217 | N | 440.1211722 | E | 4 | 15 | 0.8
$INGNS | 90214.5 | 5051.890217 | N | 440.1211722 | E | RRRN | 15 | 0.8
$INGGA | 90215 | 5051.890218 | N | 440.1211725 | E | 4 | 15 | 0.8
$INGNS | 90215 | 5051.890218 | N | 440.1211725 | E | RRRN | 15 | 0.8
$INGGA | 90215.5 | 5051.890218 | N | 440.1211719 | E | 4 | 15 | 0.8
$INGNS | 90215.5 | 5051.890218 | N | 440.1211719 | E | RRRN | 15 | 0.8
$INGGA | 90216 | 5051.890218 | N | 440.1211711 | E | 4 | 13 | 0.8
$INGNS | 90216 | 5051.890218 | N | 440.1211711 | E | RRRN | 13 | 0.8
$INGGA | 90216.5 | 5051.890218 | N | 440.1211716 | E | 4 | 15 | 0.8
$INGNS | 90216.5 | 5051.890218 | N | 440.1211716 | E | RRRN | 15 | 0.8
$INGGA | 90217 | 5051.890218 | N | 440.121173 | E | 4 | 15 | 0.8
$INGNS | 90217 | 5051.890218 | N | 440.121173 | E | RRRN | 15 | 0.8
$INGGA | 90217.5 | 5051.890218 | N | 440.1211718 | E | 4 | 14 | 0.8
$INGNS | 90217.5 | 5051.890218 | N | 440.1211718 | E | RRRN | 14 | 0.8
$INGGA | 90218 | 5051.89022 | N | 440.1211741 | E | 4 | 15 | 0.8
$INGNS | 90218 | 5051.89022 | N | 440.1211741 | E | RRRN | 15 | 0.8
$INGGA | 90218.5 | 5051.890219 | N | 440.1211735 | E | 4 | 15 | 0.8
$INGNS | 90218.5 | 5051.890219 | N | 440.1211735 | E | RRRN | 15 | 0.8
$INGGA | 90219 | 5051.890219 | N | 440.1211766 | E | 4 | 15 | 0.8
[... 200 lines cut ...]
$INGNS | 90309 | 5051.885855 | N | 440.1229571 | E | RRRN | 12 | 1.2
$INGGA | 90309.5 | 5051.885858 | N | 440.1229703 | E | 4 | 11 | 1.3
$INGNS | 90309.5 | 5051.885858 | N | 440.1229703 | E | RRRN | 11 | 1.3
$INGGA | 90310 | 5051.88589 | N | 440.1229732 | E | 4 | 12 | 0.9
$INGNS | 90310 | 5051.88589 | N | 440.1229732 | E | RRRN | 12 | 0.9
$INGGA | 90310.5 | 5051.885941 | N | 440.1229688 | E | 4 | 13 | 0.8
$INGNS | 90310.5 | 5051.885941 | N | 440.1229688 | E | RRRN | 13 | 0.8
$INGGA | 90311 | 5051.885957 | N | 440.123002 | E | 4 | 14 | 0.8
$INGNS | 90311 | 5051.885957 | N | 440.123002 | E | RRRN | 14 | 0.8
$INGGA | 90311.5 | 5051.885973 | N | 440.1230645 | E | 4 | 13 | 0.9
$INGNS | 90311.5 | 5051.885973 | N | 440.1230645 | E | RRRN | 13 | 0.9
$INGGA | 90312 | 5051.885949 | N | 440.1230796 | E | 4 | 13 | 0.9
$INGNS | 90312 | 5051.885949 | N | 440.1230796 | E | RRRN | 13 | 0.9
$INGGA | 90312.5 | 5051.885916 | N | 440.1231382 | E | 4 | 12 | 0.9
$INGNS | 90312.5 | 5051.885916 | N | 440.1231382 | E | RRRN | 12 | 0.9
$INGGA | 90313 | 5051.885925 | N | 440.1232371 | E | 4 | 11 | 1.3
$INGNS | 90313 | 5051.885925 | N | 440.1232371 | E | RRRN | 11 | 1.3
$INGGA | 90313.5 | 5051.885968 | N | 440.1233678 | E | 4 | 13 | 1
$INGNS | 90313.5 | 5051.885968 | N | 440.1233678 | E | RRRN | 13 | 1
$INGGA | 90314 | 5051.886013 | N | 440.1235141 | E | 4 | 13 | 0.9

```

I gave a quick look at how the messages are generated on the rosdriver and noticed for `GNSS` receiver the status bit is masked while this is not applied for `INS` receivers. 
How should I interpret the data on the nav rostopics?

thomasemter, 2024-07-17:
Hi @federicobenna,

the status generation indeed made no sense at all. Thank you for reporting this.

You can find a fix [here](https://github.com/thomasemter/septentrio_gnss_driver). This fix will be added to the upcoming release.

federicobenna, 2024-07-18:
Hi again @thomasemter,
Thank you for your quick reply.

The status reported is still quite incosistent across messages. I tested again using the suggested patch and got no fix on `/gpsfix` status, "normal" fix on `/navsatfix` and proper RTK fix in `GGA` msg. Receiver reported RTK fix the whole time instead ...
[image]

I suspect the issue is in the way the `last_insnavgeod_` or so for other types is parsed or updated?

thomasemter, 2024-07-18:
Sorry, the mask was missing for the status of the `/gpsfix`. It should be correct now.