septentrio-gnss/septentrio_gnss_driver #133: .nan in orientation filed of pose topic

mirons5678, 2024-07-19:
Hi

I'm using ROS 2 (Humble on Ubuntu 22.04)  with the septentrio_gnss_driver with mosaic-H and dual antenna setup. In the web gui I can see that a heading and pitch are being correctly reported. However the ROS "pose" topic contains only ".nan" for the "orientation" field.

Are you able to help please?

thomasemter, 2024-07-19:
Hi @mirons5678,

your observation is correct. For the GNSS receivers, the orientation is set to .nan on purpose. This is because only an INS is able to determine full orientation. For a dual-antenna GNSS it depends on the antenna setup which tilt angle can be observed. In case the heading offset is not a multiple of 90 °, it is neither roll nor pitch. Maybe we can think of outputting only heading?

thomasemter, 2024-07-19:
Hi @mirons5678 ,

thank you for your report. The calculation for the GNSS contained an error, since it can only determine either roll or pitch and the other is .nan. I have a quick fix [here]() and the fix will be part of the next update. Because this is valid just for a heading offset of  a multiple of 90 °, pitch and roll have to be used with care. Only an INS is able to determine full orientation.

mirons5678, 2024-07-19:
Thanks for the quick reply @thomasemter. I was hoping to fuse heading from the GNSS into a robot localization EKF node, which I think requires a Pose message with orientation - I'm using a heading offset of 90°. Should your fix enable me to do this? Do you have any idea when the next update will be released?

thomasemter, 2024-07-19:
Heading can be used for sure. If you are referring to [robot_localization](https://github.com/cra-ros-pkg/robot_localization) it is easily possible to exclude pitch and roll.

I hope the next update will be released very soon.

mirons5678, 2024-07-22:
Thanks for your support @thomasemter . With your updated branch I'm seeing orientation data in the Pose message. I'm fusing this into [robot_localization](https://github.com/cra-ros-pkg/robot_localization) and only including yaw from the GNSS pose topic. This is fused with x and y velocity from a wheel odometry topic. However, when the x and y velocity are 0, the yaw angle is continually rotating through 360° (even when the antenna are static and the web interface is showing an accurate heading). Do you have any idea why this might be?

I've also observed that setting `att_offset: heading` in the GNSS driver config file does not appear to be honoured (e.g. setting `att_offset: heading: 90.0` shows as a heading offset of 0.0 in the web GUI).

thomasemter, 2024-07-23:
Great to hear. I am sorry, but I am not very familiar with robot_localization since we use our own fusion framework at our institute.

In my setup, the heading in the WebUI reflects the parameter setting.
[image]
If you have set `get_spatial_config_from_tf: true`, the heading offset is drawn from `tf` and setting `att_offset.heading` has no effect. Also, if you have set `use_ros_axis_orientation: true`, the sign of the heading offset will be reversed since it follows a different convention.