septentrio-gnss/septentrio_gnss_driver #136: Building Package is not possible

Andyjun862, 2024-08-19:
Hello together,
I tried to launch the ROS2 Package of septrentio but I get the error message ` "file rover.py not found" ` I
System info:
Raspberry Pi 4 
OS Ubuntu 
Septentrio driver install via apt-get -- self building not a option due to crushing of Raspberry every time i do it.
Thank you in advance

thomasemter, 2024-08-20:
Hello @Andyjun862 ,
what exactly did you try to launch? Actually, the stock launch file is `rover.launch.py`. Can you give a little more detail, i.e., ROS version, driver version etc?

Are you still interested in building the package yourself? It would be helpful to have more info in order to help you. Is there a compiler error or what does "crushing" mean?

Andyjun862, 2024-08-27:
Sorry for the long waiting @thomasemter, the building on my  `Raspberry Pi 4 ` was not possible because it crushed all the time. But fortunately, I found a solution:
```
export MAKEFLAGS="-j 1"
colcon build --symlink-install --executor sequential --packages-select septentrio_gnss_driver
``` 
Maybe I will have other questions during my implementation, but I will open another Issue for this.

closed 2024-08-27 by Andyjun862

thomasemter, 2024-08-28:
Hi @Andyjun862,

I am happy you worked it out and thanx for providing the solution.

The C++ compiler got also killed sometimes for me on a quite powerful PC with colcon. And there is a discussion on ROS discourse going on where this also happens on the [build farm](https://discourse.ros.org/t/openvdb-compile-size/38790).