septentrio-gnss/septentrio_gnss_driver #143: Compile definitions not exported through cmake

oysstu, 2024-11-15:
https://github.com/septentrio-gnss/septentrio_gnss_driver/blob/5e31b44e377775261a88fa34a0ec8a246e820aac/CMakeLists.txt#L36
https://github.com/septentrio-gnss/septentrio_gnss_driver/blob/5e31b44e377775261a88fa34a0ec8a246e820aac/CMakeLists.txt#L146
https://github.com/septentrio-gnss/septentrio_gnss_driver/blob/5e31b44e377775261a88fa34a0ec8a246e820aac/CMakeLists.txt#L255

If you instead use `target_compile_definitions(<target> PUBLIC <definition>)`, I believe these should be automatically exported and defined in other projects through ament.

thomasemter, 2024-11-15:
Hi @oysstu ,

thanks for the hint. I'm not sure why we would export these, though. They are just used internally as compiler directives.

oysstu, 2024-11-15:
They are exposed through the header files. In other words, if one want to put the RosaicNode in a ROS2 executor using static composition (i.e., not through composable nodes), rosaic_node.hpp has to be included, which fails without these definitions.

thomasemter, 2024-11-15:
Ah, thank you so much for the explanation. We will change it.

closed 2025-02-18 by oysstu

oysstu, 2025-06-19:
This is still not resolved, as ament needs to be told which targets to export. You can verify this by searching for "ROS2_VER_N250" in the following folder `<workspace>/install/septentrio_gnss_driver/share/septentrio_gnss_driver/cmake/`. It should appear somewhere there associated with the target.

See the following
https://docs.ros.org/en/kilted/How-To-Guides/Ament-CMake-Documentation.html#installing

oysstu, 2025-06-19:
Our builds has completely broken with your latest changes to CMakeLists. I essentially have to add all your dependencies manually to our link target instead of relying on cmake propagating the dependencies. I've tried using the deprecated ament_target_dependencies macro and linking with various targets such as `${septentrio_gnss_driver_TARGETS}` and `${septentrio_gnss_driver_LIBRARIES}`. Nothing seems to work.

Perhaps this is resolved if you re-add the cmake exports. There should be a `septentrio_gnss_driverExport.cmake` file or something along those lines in your install directory. I also haven't seen the `ament_auto`  being used anywhere else. Personally I'd draw inspiration from the cmake setup in the link I pasted above and the core ROS2 C++ libraries (rclcpp, rclcpp_action, rclcpp_lifecycle).

thomasemter, 2025-06-22:
It seems that `ament_auto` does not export the compile definitions. So, we will revert back to normal `ament`. Can you please try with the fix in this [repo](https://github.com/thomasemter/septentrio_gnss_driver) (master branch).