septentrio-gnss/septentrio_gnss_driver #145: Hanging and trash data when both driver connection and rtk corrections through serial port

grzesiek2201, 2024-12-06:
When using one serial port for both the driver connection and ntrip corrections, the driver regularly hangs for a few seconds and doesn't communicate with the module / doesn't publish messages.
When setup is the same, but device is connected through tcp, no problems occur.

Connection:
Connected to the mosaic-go-heading with a USB cable. Using external client to provide ntrip corrections.

Config file:
```
# Configuration Settings for the Rover Rx

# GNSS/INS Parameters
septentrio_gnss_driver:
  ros__parameters:
    # device: tcp://192.168.3.1:28784
    device: serial:/dev/ttyACM0

    serial:
      baudrate: 921600
      hw_flow_control: "off"
      rx_serial_port: USB1

    stream_device:
      tcp:
        ip_server: ""
        port: 0        
      udp:
        ip_server: ""
        port: 0
        unicast_ip: ""

    configure_rx: true

    custom_commands_file: ""  

    osnma:
      mode: "off"
      ntp_server: ""
      keep_open: true

    frame_id: gnss

    imu_frame_id: imu

    poi_frame_id: base_link

    vsm_frame_id: vsm

    aux1_frame_id: aux1

    vehicle_frame_id: base_link

    local_frame_id: odom

    insert_local_frame: false

    get_spatial_config_from_tf: false

    lock_utm_zone: true

    use_ros_axis_orientation: true

    receiver_type: gnss    

    multi_antenna: true

    datum: Default

    poi_to_arp:
      delta_e: 0.0
      delta_n: 0.0
      delta_u: 0.0

    att_offset:
      heading: -90.0
      pitch: 0.0
      
    ant_type: "Unknown"
    ant_serial_nr: "Unknown"
    ant_aux1_type: "Unknown"
    ant_aux1_serial_nr: "Unknown"

    polling_period:
      pvt: 500
      rest: 500

    # time
    use_gnss_time: true
    ntp_server: false
    ptp_server_clock: false
    latency_compensation: true

    rtk_settings:
      serial_1:
        port: "USB1"
        baud_rate: 921600
        rtk_standard: "auto"
        send_gga: "auto"
        keep_open: true

    publish:
      # For both GNSS and INS Rxs
      auto_publish: false
      publish_only_valid: false
      navsatfix: true
      gpsfix: false
      gpgga: false
      gprmc: false
      gpst: false
      measepoch: false
      pvtcartesian: false
      pvtgeodetic: false
      basevectorcart: false
      basevectorgeod: false
      poscovcartesian: false
      poscovgeodetic: true
      velcovcartesian: false
      velcovgeodetic: false
      atteuler: false
      attcoveuler: false
      pose: false
      twist: false
      diagnostics: false
      aimplusstatus: false
      galauthstatus: false
      # For GNSS Rx only
      gpgsa: false
      gpgsv: false
      # For INS Rx only
      insnavcart: false
      insnavgeod: false
      extsensormeas: false
      imusetup: false
      velsensorsetup: false
      exteventinsnavcart: false
      exteventinsnavgeod: false
      imu: false
      localization: false
      tf: false

    # logger

    activate_debug_log: true

```

Debug output from the septentrio_gnss_driver:
```
gps  | [septentrio_gnss_driver_node-1] 1733471481.022090729: [septentrio_gnss_driver] [DEBUG]	AsyncManager string read fault, sync 1 found.
gps  | [septentrio_gnss_driver_node-1] 1733471481.022143883: [septentrio_gnss_driver] [DEBUG]	AsyncManager sync byte 2 read fault, should never come here.. Received byte was �
gps  | [septentrio_gnss_driver_node-1] 1733471481.022273232: [septentrio_gnss_driver] [DEBUG]	LF wo CR: A��i
gps  | [septentrio_gnss_driver_node-1] 1733471481.993846553: [septentrio_gnss_driver] [DEBUG]	AsyncManager string read fault, sync 1 found.
gps  | [septentrio_gnss_driver_node-1] 1733471482.020343574: [septentrio_gnss_driver] [DEBUG]	AsyncManager string read fault, sync 1 found.
gps  | [septentrio_gnss_driver_node-1] 1733471482.020510332: [septentrio_gnss_driver] [DEBUG]	AsyncManager sync byte 2 read fault, should never come here.. Received byte was �
gps  | [septentrio_gnss_driver_node-1] 1733471482.020963180: [septentrio_gnss_driver] [DEBUG]	LF wo CR: ��EK	����L�(@2�ʤ
gps  | [septentrio_gnss_driver_node-1] 1733471482.021265367: [septentrio_gnss_driver] [DEBUG]	handleCd: ������D�6\x��y�+�
gps  | 2024/12/06 07:51:40 [CC---]     357732 B    7608 bps (0) flepos.vlaanderen.be/FLEPOSVRS32GREC (1) /dev/ttyACM0 
gps  | [septentrio_gnss_driver_node-1] 1733471482.495656689: [septentrio_gnss_driver] [DEBUG]	AsyncManager string read fault, sync 1 found.
gps  | [septentrio_gnss_driver_node-1] 1733471483.020796817: [septentrio_gnss_driver] [DEBUG]	LF wo CR: �
gps  | [septentrio_gnss_driver_node-1] 1733471483.495114330: [septentrio_gnss_driver] [DEBUG]	AsyncManager string read fault, sync 1 found.
�@p� | [septentrio_gnss_driver_node-1] 1733471484.021693021: [septentrio_gnss_driver] [DEBUG]	LF wo CR: �Y��
gps  | [septentrio_gnss_driver_node-1] 1733471484.021693437: [septentrio_gnss_driver] [DEBUG]	handleCd: �
gps  | [septentrio_gnss_driver_node-1] 1733471485.020294326: [septentrio_gnss_driver] [DEBUG]	AsyncManager string read fault, sync 1 found.
gps  | [septentrio_gnss_driver_node-1] 1733471485.020472796: [septentrio_gnss_driver] [DEBUG]	AsyncManager sync byte 2 read fault, should never come here.. Received byte was �
gps  | [septentrio_gnss_driver_node-1] 1733471485.020947469: [septentrio_gnss_driver] [DEBUG]	LF wo CR: <�/7L܈
gps  | [septentrio_gnss_driver_node-1] 1733471485.495842316: [septentrio_gnss_driver] [DEBUG]	AsyncManager string read fault, sync 1 found.
gps  | [septentrio_gnss_driver_node-1] 1733471486.495793414: [septentrio_gnss_driver] [DEBUG]	AsyncManager string read fault, sync 1 found.
gps  | [septentrio_gnss_driver_node-1] 1733471487.021493655: [septentrio_gnss_driver] [DEBUG]	AsyncManager string read fault, sync 1 found.
gps  | [septentrio_gnss_driver_node-1] 1733471487.021686814: [septentrio_gnss_driver] [DEBUG]	AsyncManager sync byte 2 read fault, should never come here.. Received byte was  
gps  | 2024/12/06 07:51:45 [CC---]     362761 B    7600 bps (0) flepos.vlaanderen.be/FLEPOSVRS32GREC (1) /dev/ttyACM0 
gps  | [septentrio_gnss_driver_node-1] 1733471488.019894537: [septentrio_gnss_driver] [DEBUG]	handleCd: �
gps  | [septentrio_gnss_driver_node-1] 1733471488.020128498: [septentrio_gnss_driver] [DEBUG]	AsyncManager string read fault, sync 1 found.
gps  | [septentrio_gnss_driver_node-1] 1733471488.020256374: [septentrio_gnss_driver] [DEBUG]	AsyncManager sync byte 2 read fault, should never come here.. Received byte was �
gps  | [septentrio_gnss_driver_node-1] 1733471488.020423772: [septentrio_gnss_driver] [DEBUG]	LF wo CR: A��i
gps  | [septentrio_gnss_driver_node-1] 1733471488.496842564: [septentrio_gnss_driver] [DEBUG]	AsyncManager string read fault, sync 1 found.
gps  | [septentrio_gnss_driver_node-1] 1733471489.019050167: [septentrio_gnss_driver] [DEBUG]	LF wo CR: �
```

These AsyncManager errors also occur when ntrip client is turned off.

thomasemter, 2024-12-06:
Serial connections can only be used exclusively as far as I know. Can you try to use USB2 for NTRIP?

Is there a reason you do not want to use the TCP connection?

grzesiek2201, 2024-12-06:
One of the reasons I cannot use the TCP connection is that the Linux kernel on my Orin Nx doesn't include the rndis_host driver, which results in no virtual ethernet interface. But the other reason is that I have to use the com port on the Orin instead of the usb port.

thomasemter, 2024-12-06:
Ok, thanks for clarifying. I was not aware that RNDIS is not possible on Orin.

I think I have not yet fully understood your setup. Which Septentrio module are you using? Are you connected via USB or a real serial port? What is the source of your RTK corrections? Also it would be helpful to know which driver and ROS version you are using.

NTRIP via ROS does not make sense, since it is a TCPIP stream which can be received by the Septentrio directly. Maybe RTCM would be possible but then again this could be sent to the Septentrio directly.

grzesiek2201, 2024-12-06:
Unfortunately on our firmware, which is based on kernel 5.15.136, there is no redis_host, indeed. The Orin Nx is actually a Syslogic RMLA4NX16-M202S-02 computer.

The setup as it is now:
mosaic-go heading evaluation kit: usb -> Orin Nx: usb (/dev/ttyACM0)
The setup will change to:
mosaic-go heading evaluation kit: COM1 -> Orin: RS232 port (COM)

The source of RTK corrections is RTKLIB's str2str tool, via the usb serial port /dev/ttyACM0 (set up as described [here](https://github.com/septentrio-gnss/Septentrio_AgnosticCorrectionsProgram/blob/main/str2str/README.md)).

I'm using ros2 iron (humble/iron/jazzy, not much difference) and septentior_gnss_driver v1.4.1.

thomasemter, 2024-12-06:
Thanks for the details.

As I said before, it is not possible to use one serial port with two programs. Please, try to set the str2str tool to use  /dev/ttyACM1 and the driver to use USB2.

When the setup changes to RS232 port, you will have to set up a second port.

grzesiek2201, 2024-12-10:
Thank you for the support. 
I do still have a question about the debug messages 
`AsyncManager sync byte 2 read fault, should never come here.. Received byte was {`,
 `AsyncManager string read fault, sync 1 found.`, etc. 
 If I turn off (comment out in the config file) all the rtk corrections, they are still present. I suppose these are SBF blocks being sent from the receiver, but somehow they aren't correctly received/read by the ros driver? Could you provide more insight into this topic?

thomasemter, 2024-12-10:
These debug logs signal that some messages were received, which cannot be decoded by the driver. Are you still sending RTK corrections on the same serial port?

grzesiek2201, 2024-12-10:
I am not sending rtk corrections, the ntrip client is turned off. 
The log file has all the rtk settings commented out. `gpgga`, `poscovcartesian` and `navsatfix` topics are set to be published.
Here are the debug messages when starting the ros driver:
```
gps  | [septentrio_gnss_driver_node-1] 1733826617.091376314: [septentrio_gnss_driver] [DEBUG]	Called ROSaicNode() constructor..
gps  | [septentrio_gnss_driver_node-1] 1733826617.096479722: [septentrio_gnss_driver] [DEBUG]	IMU roll offset: 180.000000
gps  | [septentrio_gnss_driver_node-1] 1733826617.096576461: [septentrio_gnss_driver] [DEBUG]	IMU pitch offset: -0.000000
gps  | [septentrio_gnss_driver_node-1] 1733826617.096600909: [septentrio_gnss_driver] [DEBUG]	IMU yaw offset: -0.000000
gps  | [septentrio_gnss_driver_node-1] 1733826617.096620782: [septentrio_gnss_driver] [DEBUG]	Ant heading offset: 90.000000
gps  | [septentrio_gnss_driver_node-1] 1733826617.096639982: [septentrio_gnss_driver] [DEBUG]	Ant pitch offset: -0.000000
gps  | [septentrio_gnss_driver_node-1] 1733826617.097412671: [septentrio_gnss_driver] [DEBUG]	Finished getROSParams() method
gps  | [septentrio_gnss_driver_node-1] 1733826617.097462240: [septentrio_gnss_driver] [DEBUG]	Called connect() method
gps  | [septentrio_gnss_driver_node-1] 1733826617.097482401: [septentrio_gnss_driver] [DEBUG]	Started timer for calling connect() method until connection succeeds
gps  | [septentrio_gnss_driver_node-1] 1733826617.097520673: [septentrio_gnss_driver] [DEBUG]	Called initializeIo() method
gps  | [septentrio_gnss_driver_node-1] 1733826617.097614019: [septentrio_gnss_driver] [DEBUG]	AsyncManager created.
gps  | [septentrio_gnss_driver_node-1] 1733826617.097703045: [septentrio_gnss_driver] [INFO]	Connecting serially to device /dev/ttyACM0, targeted baudrate: 921600
gps  | [septentrio_gnss_driver_node-1] 1733826617.098504791: [septentrio_gnss_driver] [DEBUG]	Gradually increasing the baudrate to the desired value...
gps  | [septentrio_gnss_driver_node-1] 1733826617.098556120: [septentrio_gnss_driver] [DEBUG]	Initiated current_baudrate object...
gps  | [septentrio_gnss_driver_node-1] 1733826617.098582713: [septentrio_gnss_driver] [DEBUG]	Current baudrate is 921600
gps  | [septentrio_gnss_driver_node-1] 1733826617.098602137: [septentrio_gnss_driver] [INFO]	Set ASIO baudrate to 921600, leaving InitializeSerial() method
gps  | [septentrio_gnss_driver_node-1] 1733826617.098830238: [septentrio_gnss_driver] [DEBUG]	Configure Rx.
gps  | [septentrio_gnss_driver_node-1] 1733826617.098865503: [septentrio_gnss_driver] [DEBUG]	Called configureRx() method
SSSSSSSSSSptentrio_gnss_driver_node-1] 1733826617.099046659: [septentrio_gnss_driver] [DEBUG]	AsyncManager sent the following 13 bytes to the Rx: 
gps  | [septentrio_gnss_driver_node-1] 1733826617.099757907: [septentrio_gnss_driver] [DEBUG]	handleCd: USB1>
SSSSSSSSSSptentrio_gnss_driver_node-1] 1733826617.100032761: [septentrio_gnss_driver] [DEBUG]	AsyncManager sent the following 13 bytes to the Rx: 
gps  | [septentrio_gnss_driver_node-1] 1733826617.111748411: [septentrio_gnss_driver] [DEBUG]	handleCd: USB1>
SSSSSSSSSSptentrio_gnss_driver_node-1] 1733826617.111965472: [septentrio_gnss_driver] [DEBUG]	AsyncManager sent the following 13 bytes to the Rx: 
gps  | [septentrio_gnss_driver_node-1] 1733826617.112652847: [septentrio_gnss_driver] [DEBUG]	handleCd: USB1>
gps  | [septentrio_gnss_driver_node-1] 1733826617.112791506: [septentrio_gnss_driver] [INFO]	The connection descriptor is USB1
gps  | [septentrio_gnss_driver_node-1] 1733826617.112872628: [septentrio_gnss_driver] [INFO]	Setting up Rx.
gps  | [septentrio_gnss_driver_node-1] 1733826617.113000343: [septentrio_gnss_driver] [DEBUG]	AsyncManager sent the following 27 bytes to the Rx: sso, all, none, none, off 
gps  | [septentrio_gnss_driver_node-1] 1733826617.173403596: [septentrio_gnss_driver] [DEBUG]	The Rx's response contains 32 bytes and reads:
gps  | [septentrio_gnss_driver_node-1]  $R: sso, all, none, none, off 
gps  | [septentrio_gnss_driver_node-1] 
gps  | [septentrio_gnss_driver_node-1] 1733826617.173548399: [septentrio_gnss_driver] [DEBUG]	A message received:   SBFOutput, Stream1, none, none, off
gps  | [septentrio_gnss_driver_node-1] 
gps  | [septentrio_gnss_driver_node-1] 1733826617.173645009: [septentrio_gnss_driver] [DEBUG]	AsyncManager sent the following 27 bytes to the Rx: sno, all, none, none, off 
gps  | [septentrio_gnss_driver_node-1] 1733826617.173763380: [septentrio_gnss_driver] [DEBUG]	A message received:   SBFOutput, Stream2, none, none, off
gps  | [septentrio_gnss_driver_node-1] 
gps  | [septentrio_gnss_driver_node-1] 1733826617.173922647: [septentrio_gnss_driver] [DEBUG]	A message received:   SBFOutput, Stream3, none, none, off
gps  | [septentrio_gnss_driver_node-1] 
gps  | [septentrio_gnss_driver_node-1] 1733826617.174098075: [septentrio_gnss_driver] [DEBUG]	A message received:   SBFOutput, Stream4, none, none, off
gps  | [septentrio_gnss_driver_node-1] 
gps  | [septentrio_gnss_driver_node-1] 1733826617.174197182: [septentrio_gnss_driver] [DEBUG]	A message received:   SBFOutput, Stream5, none, none, off
gps  | [septentrio_gnss_driver_node-1] 
gps  | [septentrio_gnss_driver_node-1] 1733826617.174285344: [septentrio_gnss_driver] [DEBUG]	A message received:   SBFOutput, Stream6, none, none, off
gps  | [septentrio_gnss_driver_node-1] 
[... cut ...]
gps  | [septentrio_gnss_driver_node-1] 1733826618.679987661: [septentrio_gnss_driver] [DEBUG]	Leaving configureRx() method
gps  | [septentrio_gnss_driver_node-1] 1733826618.680014061: [septentrio_gnss_driver] [DEBUG]	handleCd: USB1>
gps  | [septentrio_gnss_driver_node-1] 1733826618.680200145: [septentrio_gnss_driver] [INFO]	Setup complete.
gps  | [septentrio_gnss_driver_node-1] 1733826618.680379765: [septentrio_gnss_driver] [DEBUG]	Successully connected. Leaving connect() method
gps  | [septentrio_gnss_driver_node-1] 1733826618.680547289: [septentrio_gnss_driver] [DEBUG]	Leaving ROSaicNode() constructor..
```

thomasemter, 2024-12-11:
Okay, this is strange. 

Can you please post your config or a minimal example of it. I will try to reproduce it.

grzesiek2201, 2024-12-13:
I connected the receiver the a PC where I could see the web interface, and as you said, it was the rtcm messages. They were being sent from the receiver. 
It is a bit unexpected though, as I thought that the ros driver resets all the receiver's configuration if `configure_rx : true`. Once I manually removed the rtcm output messages from the web interface, the errors disappeared. 
Did I misunderstand the `configure_rx` parameter? Or does the below config specify to output rtcm messages? However, after manually removing the rtcm output, it is not configured again when running the ros driver.

Here is the config:

```
# Configuration Settings for the Rover Rx

# GNSS/INS Parameters
septentrio_gnss_driver:
  ros__parameters:
    # device: tcp://192.168.3.1:28784
    device: serial:/dev/ttyACM0

    serial:
      baudrate: 921600
      hw_flow_control: "off"
      rx_serial_port: USB1

    stream_device:
      tcp:
        ip_server: ""
        port: 0        
      udp:
        ip_server: ""
        port: 0
        unicast_ip: ""

    configure_rx: true

    custom_commands_file: ""  

    osnma:
      mode: "off"
      ntp_server: ""
      keep_open: true

    frame_id: gnss

    imu_frame_id: imu

    poi_frame_id: base_link

    vsm_frame_id: vsm

    aux1_frame_id: aux1

    vehicle_frame_id: base_link

    local_frame_id: odom

    insert_local_frame: false

    get_spatial_config_from_tf: false

    lock_utm_zone: true

    use_ros_axis_orientation: true

    receiver_type: gnss    

    multi_antenna: true

    datum: Default

    poi_to_arp:
      delta_e: 0.0
      delta_n: 0.0
      delta_u: 0.0

    att_offset:
      heading: -90.0
      pitch: 0.0
      
    ant_type: "Unknown"
    ant_serial_nr: "Unknown"
    ant_aux1_type: "Unknown"
    ant_aux1_serial_nr: "Unknown"

    polling_period:
      pvt: 500
      rest: 500

    # time
    use_gnss_time: true
    ntp_server: false
    ptp_server_clock: false
    latency_compensation: true

    publish:
      # For both GNSS and INS Rxs
      auto_publish: false
      publish_only_valid: false
      navsatfix: true
      gpsfix: false
      gpgga: false
      gprmc: false
      gpst: false
      measepoch: false
      pvtcartesian: false
      pvtgeodetic: false
      basevectorcart: false
      basevectorgeod: false
      poscovcartesian: false
      poscovgeodetic: true
      velcovcartesian: false
      velcovgeodetic: false
      atteuler: false
      attcoveuler: false
      pose: false
      twist: false
      diagnostics: false
      aimplusstatus: false
      galauthstatus: false
      # For GNSS Rx only
      gpgsa: false
      gpgsv: false
      # For INS Rx only
      insnavcart: false
      insnavgeod: false
      extsensormeas: false
      imusetup: false
      velsensorsetup: false
      exteventinsnavcart: false
      exteventinsnavgeod: false
      imu: false
      localization: false
      tf: false

    # logger

    activate_debug_log: true
```