septentrio-gnss/septentrio_gnss_driver #149: Issue with Orientation Data in IMU Topic on origin/dev Branch

ayoubasritii, 2025-02-18:
Hello,

I am currently using the Septentrio INS Pro+ for inertial navigation, and I've encountered an issue where the orientation data within the IMU topic is empty. We are utilizing the origin/dev branch of the septentrio_gnss_driver.

I noticed a similar issue reported in your repository concerning NaN values in orientation data, which can be found here: [Issue #122](https://github.com/septentrio-gnss/septentrio_gnss_driver/issues/122). It appears that a fix was implemented for the ROS2 branch.

Could you please confirm if this fix has also been applied to the ROS1 branch(origin/dev)?

Thank you for your assistance.

thomasemter, 2025-02-18:
Hello @ayoubasritii,

yes, the fixed applies to both ROS 1 and ROS 1.

Can you please be more specific what "empty" means. Is the message published at all and if so, is it filled with zeros or NaNs? For your info: Septentrio INS only publishes IMU data after it has acquired GNSS.

ayoubasritii, 2025-02-20:
Hello @thomasemter 
Thanks for your prompt reply.
By empty I mean NAN, yes GNSS data was available and the INS was even aligned. 
Below the ros topic echo of imu topic:

> rostopic echo -n 1 imu_ins 
> header: 
>   seq: 36955
>   stamp: 
>     secs: 1739273632
>     nsecs: 488831954
>   frame_id: "tbuggy/sensor_wgs84"
> orientation: 
>   x: nan
>   y: nan
>   z: nan
>   w: nan
> orientation_covariance: [-1.0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, 0.0, -1.0]
> angular_velocity: 
>   x: -0.0036215581978882305
>   y: 0.10207994462601827
>   z: 0.05144357970253282
> angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
> linear_acceleration: 
>   x: -0.10535
>   y: 0.1586375
>   z: 10.9680375
> linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
> ---

We are using origin/dev branch and the last update was on 19 Nov 2023, are you sure this fix was applied to this branch?

thomasemter, 2025-02-20:
Hello @ayoubasritii ,

thanks for clarifying. The fix was definitely not applied to the branch origin/dev. Please, always use the master branch or install via apt.

ayoubasritii, 2025-02-21:
Hello @thomasemter,
I did a test today using the latest master branch, unfortunately I'm facing the exact issue.
You will find attached the conf.yaml file I'm using as well as a rosbag.
Thanks!

[imu_nan.zip](https://github.com/user-attachments/files/18907545/imu_nan.zip)

thomasemter, 2025-02-21:
Hello @ayoubasritii,
This is strange. In my test cases it is working as it should. 

Are you absolutely sure that the INS is aligned? The bag file you provided only contains static tf, so I cannot confirm.

Nonetheless, I have changed the behavior in case the INS is not aligned yet but has GNSS heading. It then will publish a valid orientation with roll and pitch = 0.0. You may have a look at this in PR #144.

ayoubasritii, 2025-02-24:
Hello @thomasemter 
My mistake for the empty bagfile. You will find attached a new one.
I made sure to align the INS before starting the recording. 
Let me know if the recorded topics are enough.

[2025-02-24-11-35-39.zip](https://github.com/user-attachments/files/18937971/2025-02-24-11-35-39.zip)

ayoubasritii, 2025-02-28:
Hello @thomasemter ,
Any updates?
Thanks

ayoubasritii, 2025-03-11:
Hello @thomasemter  ,
Any updates please?
Thank you

thomasemter, 2025-03-11:
Hello @ayoubasritii ,

I am so sorry for the delay. I had already written an answer but presumably forgot to push the comment button.

I still cannot reproduce your problem. Thus, I did prepare a debug branch [/dev/debug_imu](https://github.com/thomasemter/septentrio_gnss_driver/tree/dev/debug_imu). Can you please try this with `activate_debug_log: true` and post the console output?

thomasemter, 2025-03-12:
Hello @ayoubasritii,

I am glad it is working for you now.

Covariance is indeed not set for angular velocity and linear acceleration. We could consider to fill them according to the data sheet of the IMU.

Which field do you address by "angles data" exactly?

Regarding the covariance ot the fix: this is calculated within the receiver and the driver just passes it on. For questions about the firmware, the [official support}(https://www.septentrio.com/en/support) is better suited.

ayoubasritii, 2025-03-13:
Hello @thomasemter 
By angles data I meant orientation.
I was getting zero values on the covariance before the alignment, is that normal?
After the alignment, the covariance seems too optimistic:
Roll SD: ≈ 0.36°
Pitch SD: ≈ 0.36°
Yaw SD: ≈ 0.46°
Are those also calculated by the INS?

thomasemter, 2025-03-13:
Hallo @ayoubasritii,

thanks for clarifying. Is the orientation always 0? That should not be the case. Could you please provide another bag file?

The covariance is plausible in my opinion. As I said, all these values are calculated inside the INS.

ayoubasritii, 2025-03-14:
Hello @thomasemter ,
My mistake, the orientation covariance value are set to -1 before alignment not zero, I think this is intended.
You said "Covariance is indeed not set for angular velocity and linear acceleration. We could consider to fill them according to the data sheet of the IMU.", What should I do exactly to fill them?

thomasemter, 2025-03-14:
Hello @ayoubasritii ,

Yes, this is intentional.

The IMU message is filled [here](https://github.com/septentrio-gnss/septentrio_gnss_driver/blob/2d91a27f711a29455df3b4cf79ac8fb87f0209f2/src/septentrio_gnss_driver/communication/message_handler.cpp#L576). The covariance values are static and can be taken from the data sheet (I am not sure but I believe it is an [ADIS16505-2](https://www.analog.com/media/en/technical-documentation/data-sheets/adis16505.pdf)) or determined with a tool like [this](https://github.com/ori-drs/allan_variance_ros).

ayoubasritii, 2025-03-20:
Hello @thomasemter ,
Are you sure the function assembleImu() does assign values to msg.angular_velocity_covariance?
It sets msg.angular_velocity values but does not assign a covariance matrix for it. Unlike msg.orientation_covariance, which gets values from last_insnavgeod_, there is no similar block for angular velocity covariance.

thomasemter, 2025-03-20:
Hallo @ayoubasritii,

sorry for the misunderstanding. The covariances are not filled. I meant to give you a hint, where you could add your own code to fill them if needed.

ayoubasritii, 2025-03-21:
Hello @thomasemter ,
I was able to reproduce the NAN orientation in imu message, I have checkout the debug branch and activated the log param as requested. You will also find attached a rosbag file.

[septentrio_bag2_2025-03-21-11-38-06.zip](https://github.com/user-attachments/files/19387230/septentrio_bag2_2025-03-21-11-38-06.zip)

[33m[ WARN] [1742542516.393001966]: /tbuggy/septentrio: Waiting for transform from tbuggy/sensor_wgs84 to tbuggy/base_link: "tbuggy/base_link" passed to lookupTransform argument target_frame does not exist. .[0m
[31m[ERROR] [1742542516.744310690]: /tbuggy/septentrio: no ins_vsm.ip_server.id set -> VSM input will not be used![0m
[32m[DEBUG] [1742542512.324352853]: /tbuggy/septentrio: Called ROSaicNode() constructor..[0m
[32m[DEBUG] [1742542516.714578314]: /tbuggy/septentrio: IMU roll offset: 180.000000[0m
[32m[DEBUG] [1742542516.714641313]: /tbuggy/septentrio: IMU pitch offset: 0.000000[0m
[32m[DEBUG] [1742542516.714660399]: /tbuggy/septentrio: IMU yaw offset: -0.000000[0m
[32m[DEBUG] [1742542516.714675175]: /tbuggy/septentrio: Ant heading offset: -0.000000[0m
[32m[DEBUG] [1742542516.714689237]: /tbuggy/septentrio: Ant pitch offset: 3.070539[0m
[32m[DEBUG] [1742542516.747616616]: /tbuggy/septentrio: Finished getROSParams() method[0m
[32m[DEBUG] [1742542516.747771170]: /tbuggy/septentrio: Leaving ROSaicNode() constructor..[0m
[32m[DEBUG] [1742542516.747896368]: /tbuggy/septentrio: Called connect() method[0m
[32m[DEBUG] [1742542516.747945217]: /tbuggy/septentrio: Started timer for calling connect() method until connection succeeds[0m
[32m[DEBUG] [1742542516.747983251]: /tbuggy/septentrio: Called initializeIo() method[0m
[32m[DEBUG] [1742542516.748127190]: /tbuggy/septentrio: AsyncManager created.[0m
[0m[ INFO] [1742542516.748318247]: /tbuggy/septentrio: Connecting to tcp://192.168.5.200:28784...[0m
[0m[ INFO] [1742542516.749951927]: /tbuggy/septentrio: Connected to 192.168.5.200:28784.[0m
[0m[ INFO] [1742542516.750169435]: /tbuggy/septentrio: Setup complete.[0m
[32m[DEBUG] [1742542516.750204196]: /tbuggy/septentrio: Successully connected. Leaving connect() method[0m
[32m[DEBUG] [1742542516.772450116]: /tbuggy/septentrio: handleCd: IP10>[0m
[32m[DEBUG] [1742542516.788698879]: Trying to publish message of type [septentrio_gnss_driver/ExtSensorMeas/bc4155d6dac83ebf51ac1aba432000ae] on a publisher with type [septentrio_gnss_driver/ExtSensorMeas/bc4155d6dac83ebf51ac1aba432000ae][0m
[32m[DEBUG] [1742542516.788755214]: /tbuggy/septentrio: last_insnavgeod_.block_header.tow: 4294967295[0m
[32m[DEBUG] [1742542516.790150557]: Trying to publish message of type [sensor_msgs/Imu/6a62c6daae103f4ff57a132d6f95cec2] on a publisher with type [sensor_msgs/Imu/6a62c6daae103f4ff57a132d6f95cec2][0m
[32m[DEBUG] [1742542516.791448421]: Trying to publish message of type [septentrio_gnss_driver/INSNavGeod/7e95cd9c24ef355a547323f4ca3d6e14] on a publisher with type [septentrio_gnss_driver/INSNavGeod/7e95cd9c24ef355a547323f4ca3d6e14][0m
[32m[DEBUG] [1742542516.792788438]: Trying to publish message of type [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7] on a publisher with type [nav_msgs/Odometry/cd5e73d190d741a2f92e81eda573aca7][0m
[32m[DEBUG] [1742542516.807331741]: /tbuggy/septentrio: tsImu: 1742542534740000000, tsIns: 1742542534720000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542516.810073335]: Trying to publish message of type [sensor_msgs/NavSatFix/2d3a8cd499b9b4a0249fb98fd05cfa48] on a publisher with type [sensor_msgs/NavSatFix/2d3a8cd499b9b4a0249fb98fd05cfa48][0m
[32m[DEBUG] [1742542516.827349239]: /tbuggy/septentrio: tsImu: 1742542534760000000, tsIns: 1742542534740000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542516.847615984]: /tbuggy/septentrio: tsImu: 1742542534780000000, tsIns: 1742542534760000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542516.868355775]: /tbuggy/septentrio: tsImu: 1742542534800000000, tsIns: 1742542534780000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542516.887275364]: /tbuggy/septentrio: tsImu: 1742542534820000000, tsIns: 1742542534800000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542516.907224048]: /tbuggy/septentrio: tsImu: 1742542534840000000, tsIns: 1742542534820000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542516.927104278]: /tbuggy/septentrio: tsImu: 1742542534860000000, tsIns: 1742542534840000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542516.947361091]: /tbuggy/septentrio: tsImu: 1742542534880000000, tsIns: 1742542534860000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542516.968012748]: /tbuggy/septentrio: tsImu: 1742542534900000000, tsIns: 1742542534880000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542516.987111552]: /tbuggy/septentrio: tsImu: 1742542534920000000, tsIns: 1742542534900000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542517.007127624]: /tbuggy/septentrio: tsImu: 1742542534940000000, tsIns: 1742542534920000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542517.026896227]: /tbuggy/septentrio: tsImu: 1742542534960000000, tsIns: 1742542534940000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542517.046591285]: /tbuggy/septentrio: tsImu: 1742542534980000000, tsIns: 1742542534960000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542517.068023638]: /tbuggy/septentrio: tsImu: 1742542535000000000, tsIns: 1742542534980000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542517.086655226]: /tbuggy/septentrio: tsImu: 1742542535020000000, tsIns: 1742542535000000000, maxDt: 10000000[0m
[... cut ...]
[32m[DEBUG] [1742542518.151756947]: Trying to publish message of type [diagnostic_msgs/DiagnosticArray/60810da900de1dd6ddd437c3503511da] on a publisher with type [diagnostic_msgs/DiagnosticArray/60810da900de1dd6ddd437c3503511da][0m
[32m[DEBUG] [1742542518.161392062]: /tbuggy/septentrio: tsImu: 1742542536100000000, tsIns: 1742542536080000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542518.180312365]: /tbuggy/septentrio: tsImu: 1742542536120000000, tsIns: 1742542536100000000, maxDt: 10000000[0m
[32m[DEBUG] [1742542518.200475005]: /tbuggy/septentrio: tsImu: 1742542536140000000, tsIns: 1742542536120000000, maxDt: 10000000[0m

thomasemter, 2025-03-21:
Thanks, I will look into it.

thomasemter, 2025-03-21:
The bag file you provided just contains some unrelated odometry messages. At least the imu topic would be helpful.

According to the debug output, the timestamps of the INS and IMU are too far apart (20 ms) and thus the orientation is discarded. Since I cannot reproduce this on my INS I do not know what is going on right now. Do you have the latest firmware (1.4.4)? Which INS do have exactly?

ayoubasritii, 2025-03-24:
Hello,

Sorry again about the bagfile, the right one is still in the robot I will try to send it tomorrow.
Yes I'm using the latest firmware 1.4.4, we are using the INS Sbi Pro+

ayoubasritii, 2025-03-25:
Hello @thomasemter ,

You will attached the rosbag, sorry for the inconvenience.

[septentrio_bag_2025-03-21-11-35-18.zip](https://github.com/user-attachments/files/19441960/septentrio_bag_2025-03-21-11-35-18.zip)

thomasemter, 2025-03-25:
Thanks for the bag file @ayoubasritii.

There is something strange going on. The rates do not match your config. Setting `polling_period.pvt: 0`, you should have 200 Hz IMU and 100 Hz INS but both are only 50 Hz.

thomasemter, 2025-03-25:
Hello @ayoubasritii ,

this was indeed an oversight. Thank you for revealing it and sorry for the inconvenience.

I have reworked the IMU publishing part. You may find it [here](https://github.com/thomasemter/septentrio_gnss_driver) if you like to test it.

ayoubasritii, 2025-03-27:
Hello @thomasemter 

I tested yesterday the fix in 2 platforms and I confirm it resolve the issue, thank you very much for you prompt responses!
Not sure if it's related, but the topic "odometry_vsm" is no longer available when doing rosnode info of the driver. The conf mentioned below should be taken into account even if `configure_rx: false` right?

> ins_vsm:
>   ros:
>     source: "odometry"
>     config: [true, true, true]
>     variances_by_parameter: true
>     variances: [0.01, 0.0001, 0.0001]

ayoubasritii, 2025-03-28:
Hello @thomasemter 
So I have to edit the ip_server even if I just want to share the vsm over ros topic?

I will be off next week so I will test this fix the week after, I think you can close this ticket for now, if needed I will create another one.
Thank you again for everything :)

thomasemter, 2025-03-28:
`odometry_vsm` is received by the driver via ROS and forwarded to the INS. Therefore, an IP server is needed on the INS side.

You are very welcome.