septentrio-gnss/septentrio_gnss_driver #157: How to publish both GNSS antenna position and INS (IMU or Vehicle) position simultaneously?

LimHaeryong, 2025-05-21:
Hi, I'm using the sbi3 pro +.

I would like to receive two separate position outputs simultaneously via ROS topics:

1. The **GNSS (main) antenna position**, based on GNSS-only solution (typically from the `PVTGeodetic` SBF block).
2. The **INS position**, based on the fused INS/GNSS solution (from the `INSNavGeod` SBF block).

According to the README.md of this package, when INS is active, the PVTGeodetic block corresponds to the INSNavGeod SBF block, rather than the original GNSS-only PVTGeodetic. As a result, it becomes difficult to distinguish between the GNSS antenna position and the INS-based position in the ROS output (/pvtgeodetic and /insnavgeod topics).

Is there any configuration or workaround to:
- Force the `PVTGeodetic` block to continue outputting the GNSS-only position even when INS is active?
- Or alternatively, can the driver differentiate and publish the GNSS-only and INS positions on separate topics?

thomasemter, 2025-05-21:
Hi @LimHaeryong,

`PVTGeodetic` and `INSNavGeod` are 1:1 copies of the data coming from the SBi. These are GNSS position and INS position respectively, just as you requested.

The statement in the README is wrong. Thank you for pointing it out und sorry for the inconvenience. We will fix it.

LimHaeryong, 2025-05-21:
Thank you for answering so quickly!

closed 2025-05-21 by LimHaeryong